PX4/PX4-Autopilot #4685: SITL gazebo not starting in master

julianoes, 2016-05-29:
Looks like another regression:

```
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = sdf::Element; typename boost::detail::sp_member_access<T>::type = sdf::Element*]: Assertion `px != 0' failed.
[Err] [ConnectionManager.cc:118] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:83] Unable to initialize transport.
[Err] [gazebo_client.cc:62] Unable to setup Gazebo
```

julianoes, 2016-05-29:
`git bisect` points at 959c9e3817685f4313fce4dbf23a0136dd5cb258

LorenzMeier, 2016-05-29:
I doubt its that. Because if you look at what changed its entirely connection unrelated:
https://github.com/PX4/sitl_gazebo/commit/f1a66579a2a25d162886f6dd7b1c02f26eba1109

I much rather think you're running into a race that is present in the current setup already and we need to debug and fix it.

LorenzMeier, 2016-05-29:
I did a stylistic change which might help you: https://github.com/PX4/sitl_gazebo/commit/7096721ff4c0314ebeee4cb1f24a84e88f2e8c9f

Can you re-test master?

julianoes, 2016-05-29:
Still same problem with latest master. I'll have to dig into what error actually happens and why.

LorenzMeier, 2016-05-29:
What is your Gazebo version? I have:

```
Gazebo multi-robot simulator, version 6.5.1
Copyright (C) 2012-2015 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

julianoes, 2016-05-30:
```
gazebo --version
Gazebo multi-robot simulator, version 6.1.0
Copyright (C) 2012-2015 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

That's Ubuntu 15.10.

LorenzMeier, 2016-05-30:
Is there a way for you to update Gazebo or is this the current released version?

LorenzMeier, 2016-05-30:
You will want to have 6.5ish.

julianoes, 2016-05-30:
I installed 7.0 from http://gazebosim.org/tutorials?cat=install&tut=install_ubuntu&ver=7.0 and that works. Is it ok to update the [devguide](http://dev.px4.io/simulation-gazebo.html#normal-installation) then?

LorenzMeier, 2016-05-30:
Yes please!

julianoes, 2016-05-30:
https://github.com/PX4/Devguide/pull/167

pickledgator, 2016-05-30:
Can confirm brew install of gazebo7 on OSX works fine on master as well (currently v7.0.0). [simulation-gazebo.html](http://dev.px4.io/simulation-gazebo.html) still references 6, though the git [readme](https://github.com/PX4/sitl_gazebo) says 7.

vjon, 2017-04-03:
The developer documentation page for Gazebo simulation has been updated for Gazebo v7.

I also set up the PX4 development environment/toolchain from scratch using the current documentation, and Gazebo SITL works fine. So I guess this issue can be closed?

mhkabir, 2017-04-04:
Yup.

closed 2017-04-04 by mhkabir

F-LAB-Systems, 2021-02-14:
[Err] [ConnectionManager.cc:109] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo

I have this  issue, what to do

Jaeyoung-Lim, 2021-02-14:
@F-LAB-Systems Please post a new issue, this issue is not relevant anymore

eastman-bechamo, 2022-06-16:
@F-LAB-Systems Did you post a new issue? I'm having the same problem

julianoes, 2022-06-19:
@F-LAB-Systems don't necrobump, open a new issue.