PX4/PX4-Autopilot #4695: master: uninformative error message "Not ready to fly: Sensors not set up correctly"

kd0aij, 2016-05-30:
QGC shows all sensors "green". But there is not even a hint as to the cause of the problem.

tops4u, 2016-05-31:
I had the same issue when connecting or disconnecting an additional Compass it would complain but Sensors all show ok in Summary Screen.

whyscience, 2016-05-31:
yeah ,the same problem

dagar, 2016-05-31:
Is this only on master or current stable? Were you able to get around it @kd0aij?

kd0aij, 2016-05-31:
No, I guess I'll have to modify commander to figure out what's wrong.

kd0aij, 2016-05-31:
commander is coded in such a way that it is extremely reluctant to report the cause of preflight check failures. Running "commander check" at the console will tell you something though.

kd0aij, 2016-05-31:
a complete sensor recalibration was required to resolve the issue

priseborough, 2016-06-01:
I had the same symptons which were resolved by repeating the magnetometer calibration, after noticing that the calibration data was not complete with scale factors missing for the primary HMC5883. This was on a Solo with PX2 autopilot.

LorenzMeier, 2016-06-21:
@julianoes @kd0aij @tumbili Time to fix this - have any of you looked into this already? I think we need to break it down into a per-sensor message.

kd0aij, 2016-06-21:
Any ideas as to why a complete sensor recalibration might be required, instead of loading a saved parameter file?

tstellanova, 2016-06-24:
Ran into this issue with latest PX4 stable firmware today. Eventually cleared after multiple (5?) mag recalibrations

mbrandonhurd, 2016-07-02:
I'm having the same issue. Any workarounds that have been found would be greatly appreciated!

julianoes, 2016-07-13:
Improved in #5032.

Pedro-Roque, 2016-07-15:
Still having the same problem. Full calibration and after some reboots still have it.

julianoes, 2016-07-15:
@Pedro-Roque after connecting, close QGroundControl and start it again and look at all the messages you receive. It should tell you something.

Pedro-Roque, 2016-07-15:
Got Accel0 and Gyro0 errors,  saying it's uncalibrated after calibration and reboot. Some times the error does not come up,  after several reboots. Others it does. Suspect a problem on the Hardware. What do you think?

julianoes, 2016-07-16:
@Pedro-Roque When it says "uncalibrated", can you check fetch the params in QGC, save it to file and upload it here please? Thanks.

Pedro-Roque, 2016-07-18:
@julianoes Here goes the link to the params file:
https://drive.google.com/open?id=0B2sq_Zd23H8WTm5Bb1Y1cnFTSG8

julianoes, 2016-07-19:
@Pedro-Roque what hardware is this? I see that you have only one Gyro and one Mag? Is it a Pixracer?

```
1	1	CAL_ACC0_ID	1114634	6
1	1	CAL_ACC0_XOFF	1.064	9
1	1	CAL_ACC0_XSCALE	1.010	9
1	1	CAL_ACC0_YOFF	1.246	9
1	1	CAL_ACC0_YSCALE	1.023	9
1	1	CAL_ACC0_ZOFF	1.307	9
1	1	CAL_ACC0_ZSCALE	1.008	9
1	1	CAL_ACC1_ID	1114634	6
1	1	CAL_ACC2_ID	0	6
1	1	CAL_ACC_PRIME	1114634	6
1	1	CAL_BARO_PRIME	0	6
1	1	CAL_GYRO0_ID	2228490	6
1	1	CAL_GYRO0_XOFF	0.005	9
1	1	CAL_GYRO0_XSCALE	1.000	9
1	1	CAL_GYRO0_YOFF	0.006	9
1	1	CAL_GYRO0_YSCALE	1.000	9
1	1	CAL_GYRO0_ZOFF	-0.007	9
1	1	CAL_GYRO0_ZSCALE	1.000	9
```

Looking at the params, I see that the `ACC0_{X,Y,Z}OFF are quite big. They are usually around < 1.

Pedro-Roque, 2016-07-19:
No, actually this is a 3DR Pixhawk unit. Hardware fault then?

Pedro-Roque, 2016-07-19:
@julianoes

julianoes, 2016-07-19:
I would expect a Pixhawk to have 2 accelerometers and 2 gyros.

Can you do the commands below in NuttShell, each of them twice.
You get the NuttShell, by ejecting the MicroSD card out of the Pixhawk and connecting it to USB, more instructions would be in http://dev.px4.io/advanced-system-console.html#opening-the-console.

```
mpu6000 test
l3gd20 test
lsm303d test
```

Pedro-Roque, 2016-07-19:
Here are the results:

`nsh> lsm303d test
WARN  [lsm303d] accel x:          0.07518       m/s^2
WARN  [lsm303d] accel y:         -1.16409       m/s^2
WARN  [lsm303d] accel z:         -9.72859       m/s^2
WARN  [lsm303d] accel x:        486     raw
WARN  [lsm303d] accel y:        21      raw
WARN  [lsm303d] accel z:        -3475   raw
WARN  [lsm303d] accel range:  78.0000 m/s^2
WARN  [lsm303d] accel antialias filter bandwidth: 30 Hz
WARN  [lsm303d] mag device active: onboard
WARN  [lsm303d] mag x:   -0.08480       ga
WARN  [lsm303d] mag y:    0.41937       ga
WARN  [lsm303d] mag z:    0.14730       ga
WARN  [lsm303d] mag x:  -1195   raw
WARN  [lsm303d] mag y:  2669    raw
WARN  [lsm303d] mag z:  -1010   raw
WARN  [lsm303d] mag range:   2.0000 ga
nsh> 
`
`nsh> l3gd20 test
WARN  [l3gd20] gyro x:   -0.01039       rad/s
WARN  [l3gd20] gyro y:   -0.00968       rad/s
WARN  [l3gd20] gyro z:   -0.00343       rad/s
WARN  [l3gd20] temp:    34      C
WARN  [l3gd20] gyro x:  -27     raw
WARN  [l3gd20] gyro y:  -1      raw
WARN  [l3gd20] gyro z:  2       raw
WARN  [l3gd20] temp:    6       raw
WARN  [l3gd20] gyro range:  34.9066 rad/s (2000 deg/s)
ERROR [l3gd20] PASS
nsh> `

`nsh> 
nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> mpu6000 test
WARN  [mpu6000] ret: -1, expected: 72
ERROR [mpu6000] immediate acc read failed
ERROR [mpu6000] Task exited with errno=22

nsh> `

MPU6000 never gets a reading. Is this a hardware failure that I can fix with some heat?

Pedro-Roque, 2016-07-20:
@julianoes

julianoes, 2016-07-20:
@Pedro-Roque Yes that looks like a hardware failure then. If the sensor fails consistently and you calibrate using just the other sensors, you should be ok though. If it starts up intermittently, you will run into trouble.

closed 2016-07-20 by julianoes

Pedro-Roque, 2016-07-20:
That's the problem,  it starts up intermittently. Do you think that changing the mpu might do it?

julianoes, 2016-07-20:
@Pedro-Roque I can only guess, maybe? :)

Darkus3, 2017-05-11:
Hi I knoy it's an old topic but I have the same issues but with my sensorr getting some value

nsh> mpu6000 test
WARN  [mpu6000] single read
WARN  [mpu6000] time:     118732626
WARN  [mpu6000] acc  x:          -0.0252        m/s^2
WARN  [mpu6000] acc  y:          -0.1929        m/s^2
WARN  [mpu6000] acc  z:         -10.3069        m/s^2
WARN  [mpu6000] acc  x:         -6      raw 0xfffa
WARN  [mpu6000] acc  y:         -84     raw 0xffac
WARN  [mpu6000] acc  z:         -4288   raw 0xef40
WARN  [mpu6000] acc range:  78.4532 m/s^2 (  8.0000 g)
WARN  [mpu6000] gyro x:          -0.02141       rad/s
WARN  [mpu6000] gyro y:           0.03093       rad/s
WARN  [mpu6000] gyro z:          -0.01119       rad/s
WARN  [mpu6000] gyro x:         -20     raw
WARN  [mpu6000] gyro y:         29      raw
WARN  [mpu6000] gyro z:         -11     raw
WARN  [mpu6000] gyro range:  34.9066 rad/s (2000 deg/s)
WARN  [mpu6000] temp:    45.5457ERROR [sensors] Accel #0 fail:          deg celsius
TOUT!
WARN  [mpu6000] temp:   3807    rawERROR [sensors] Gyro #0 fail:  T 0xedf
OUT!

LorenzMeier, 2017-05-11:
I believe this is fixed. Is this still an issue for you?

Darkus3, 2017-05-11:
Yeah, after recalibrating sensors I still got an error:

nsh> mpu6000 test
WARN  [mpu6000] single read
WARN  [mpu6000] time:     35485808
WARN  [mpu6000] acc  x:           0.0839        m/s^2
WARN  [mpu6000] acc  y:          -0.1493        m/s^2
WARN  [mpu6000] acc  z:          -9.7185        m/s^2
WARN  [mpu6000] acc  x:         -63     raw 0xffc1
WARN  [mpu6000] acc  y:         -108    raw 0xff94
WARN  [mpu6000] acc  z:         -4306   raw 0xef2e
WARN  [mpu6000] acc range:  78.4532 m/s^2 (  8.0000 g)
WARN  [mpu6000] gyro x:           0.00001       rad/s
WARN  [mpu6000] gyro y:           0.00066       rad/s
WARN  [mpu6000] gyro z:           0.00093       rad/s
WARN  [mpu6000] gyro x:         -18     raw
WARN  [mpu6000] gyro y:         31      raw
WARN  [mpu6000] gyro z:         -11     raw
WARN  [mpu6000] gyro range:  34.9066 rad/s (2000 deg/s)
WARN  [mpu6000] temp:    41.4986ERROR [sensors] Accel #0 failove        deg celsius
r : Data timeout!
WARN  [mpu6000] temp:   2346    rawERROR [sensors] Gyro #0 failover 0x92a
 : Data timeout!
nsh>