PX4/PX4-Autopilot #4705: TakeOff via MAVLink command: Altitude command param vs. vehicle parameter

ecmnet, 2016-05-31:
Sending the MAVLink command MAV_CMD_NAV_TAKEOFF with param7 (altitude set point), it seems not to be clear which altitude should be the target. Although the SP is set to value provided with the command, takeoff takes place considering the global vehicle parameter MIS_TAKEOFF_ALT in the first phase. Then, after some time, the SP provided is considered and the vehicle adjusts its altitude accordingly. This could result in a massive overshoot. Setting MIS_TAKEOFF_ALT to 0.0 solves that issue and the param7 of the command is considered as TakeOff-Altitude.

Example: MIS_TAKEOFF_ALT is set to 10m; Param7 is set to ground altitude + 1:

[image]

ecmnet, 2016-06-03:
@jgoppert This seems to be LPE related. EKF2 SITL does not show this effect at all. Any idea?

jgoppert, 2016-06-03:
I thought this is expected behavior.  Mis takeoff alt is the min takeoff alr, so it gets to that within nav accept radius of 1m, then follows mission alt, not sure how the estimators could be different here.

ecmnet, 2016-06-03:
@jgoppert Thanks for clarification. I'll re-check my SITL settings with EKF2.

ecmnet, 2016-06-03:
Ok. That's explaining it, MIS_TAKEOFF_ALT was set to 2.5m compared to 10.0 for LPE SITL. That means, after reaching 1.5m mission altitude was taken into account. Thanks again for the clarification. I'll close this issue.

closed 2016-06-03 by ecmnet

jgoppert, 2016-06-03:
We should probably change the default for lpe sitl so others are not confused as well.