PX4/PX4-Autopilot #4706: Motors won't spin after Arming

tops4u, 2016-05-31:
Today I wanted to Fly my Hexa. Again it denied spinning the motors after Arming. Strangely it attached the Messages to an existing Message Log (several Days old):

`2000_01_01_00_00_04: REJECT AUTO RTL
2016_05_20_20_07_21: ARMED by RC
2016_05_20_20_07_21: TAKEOFF DETECTED
2016_05_20_20_07_21: [blackbox] recording: 20_07_21.px4log
2016_05_20_20_07_23: DISARMED by RC
2016_05_20_20_07_23: LANDING DETECTED
2016_05_20_20_07_24: [blackbox] stopped (0 drops)
2016_05_20_20_16_34: MANUAL KILL SWITCH ENGAGED
2016_05_20_20_16_35: MANUAL KILL SWITCH OFF
2016_05_20_20_16_37: MANUAL KILL SWITCH ENGAGED
2016_05_20_20_16_38: MANUAL KILL SWITCH OFF
2016_05_20_20_16_46: [blackbox] /fs/microsd/log/2016-05-20
2016_05_20_20_16_46: TAKEOFF DETECTED
2016_05_20_20_16_47: [blackbox] recording: 20_16_46.px4log
2016_05_20_20_16_53: LANDING DETECTED
2016_05_20_20_16_54: TAKEOFF DETECTED
2016_05_20_20_17_09: DISARMED by RC
2016_05_20_20_17_09: LANDING DETECTED
2016_05_20_20_17_10: [blackbox] stopped (1334 drops)
2016_05_20_20_18_15: LOW BATTERY, TAKEOFF DISCOURAGED
2016_05_20_20_18_15: CRITICAL BATTERY, SHUT SYSTEM DOWN
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_00_43: data link #0 lost
2000_01_01_00_00_04: DISARMED by RC
2000_01_01_00_00_04: DISARMED by RC
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_00_04: DISARMED by RC
2000_01_01_00_00_23: REJECT AUTO MISSION
2000_01_01_00_00_23: REJECT AUTO MISSION
2000_01_01_00_00_24: REJECT AUTO MISSION
2000_01_01_00_00_25: REJECT AUTO MISSION
2000_01_01_00_00_25: REJECT AUTO MISSION
2000_01_01_00_00_26: REJECT AUTO MISSION
2000_01_01_00_00_27: REJECT AUTO MISSION
2000_01_01_00_00_27: REJECT AUTO MISSION
2000_01_01_00_00_28: REJECT AUTO MISSION
2000_01_01_00_00_32: REJECT AUTO MISSION
2000_01_01_00_00_33: REJECT AUTO MISSION
2000_01_01_00_00_34: REJECT AUTO MISSION
2000_01_01_00_00_34: REJECT AUTO MISSION
2000_01_01_00_00_35: REJECT POSITION CONTROL
2000_01_01_00_00_36: REJECT POSITION CONTROL
2000_01_01_00_00_36: REJECT POSITION CONTROL
2000_01_01_00_00_37: REJECT POSITION CONTROL
2000_01_01_00_00_39: REJECT POSITION CONTROL
2000_01_01_00_00_42: REJECT AUTO PAUSE
2000_01_01_00_03_21: REJECT AUTO RTL
2000_01_01_00_04_58: REJECT AUTO RTL
2000_01_01_00_05_14: REJECT AUTO RTL
2000_01_01_00_06_03: REJECT AUTO RTL
2000_01_01_00_06_37: REJECT AUTO HOLD
2000_01_01_00_06_37: REJECT AUTO HOLD
2000_01_01_00_06_38: REJECT AUTO HOLD
2000_01_01_00_06_39: REJECT AUTO HOLD
2000_01_01_00_06_39: REJECT AUTO HOLD
2000_01_01_00_06_40: REJECT AUTO HOLD
2000_01_01_00_06_40: REJECT AUTO HOLD
2000_01_01_00_06_41: REJECT AUTO HOLD
2000_01_01_00_06_41: REJECT AUTO HOLD
2000_01_01_00_06_42: REJECT AUTO HOLD
2000_01_01_00_06_42: REJECT AUTO HOLD
2000_01_01_00_06_43: REJECT AUTO HOLD
2000_01_01_00_06_43: REJECT AUTO HOLD
2000_01_01_00_06_44: REJECT AUTO HOLD
2000_01_01_00_06_45: REJECT AUTO HOLD
2000_01_01_00_06_45: REJECT AUTO HOLD
2000_01_01_00_06_46: REJECT AUTO HOLD
2000_01_01_00_06_46: REJECT AUTO HOLD
2000_01_01_00_06_47: REJECT AUTO HOLD
2000_01_01_00_06_47: REJECT AUTO HOLD
2000_01_01_00_06_48: REJECT AUTO HOLD
2000_01_01_00_06_49: REJECT AUTO HOLD
2000_01_01_00_06_49: REJECT AUTO HOLD
2000_01_01_00_06_50: REJECT AUTO HOLD
2000_01_01_00_06_51: REJECT AUTO HOLD
2000_01_01_00_06_51: REJECT AUTO HOLD
2000_01_01_00_06_52: REJECT AUTO HOLD
2000_01_01_00_06_53: REJECT AUTO HOLD
2000_01_01_00_06_56: REJECT AUTO RTL
2000_01_01_00_07_09: REJECT AUTO RTL
2000_01_01_00_07_32: REJECT AUTO RTL
2000_01_01_00_07_34: REJECT AUTO HOLD
2000_01_01_00_07_34: REJECT AUTO HOLD
2016_05_20_21_28_19: REJECT AUTO RTL
2016_05_20_21_28_29: REJECT AUTO RTL
2016_05_20_21_30_12: REJECT AUTO RTL
2016_05_20_21_32_07: data link #0 lost
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_00_35: REJECT AUTO HOLD
2000_01_01_00_00_41: REJECT AUTO HOLD
2000_01_01_00_00_47: REJECT AUTO RTL
2000_01_01_00_00_54: REJECT AUTO RTL
2000_01_01_00_03_27: REJECT AUTO HOLD
2000_01_01_00_03_43: data link #0 lost
2000_01_01_00_00_04: DISARMED by RC
2000_01_01_00_00_15: REJECT AUTO RTL
2000_01_01_00_01_17: REJECT AUTO RTL
2000_01_01_00_01_20: REJECT AUTO HOLD
2000_01_01_00_01_20: REJECT AUTO HOLD
2000_01_01_00_01_21: REJECT AUTO HOLD
2000_01_01_00_02_37: REJECT AUTO RTL
2000_01_01_00_02_53: REJECT POSITION CONTROL
2000_01_01_00_03_02: REJECT AUTO RTL
2000_01_01_00_03_04: REJECT POSITION CONTROL
2000_01_01_00_03_22: REJECT AUTO HOLD
2000_01_01_00_03_22: REJECT AUTO HOLD
2000_01_01_00_03_23: REJECT AUTO HOLD
2000_01_01_00_03_24: REJECT AUTO HOLD
2000_01_01_00_03_24: REJECT AUTO HOLD
2000_01_01_00_03_25: REJECT AUTO HOLD
2000_01_01_00_03_25: REJECT AUTO HOLD
2000_01_01_00_03_27: REJECT POSITION CONTROL
2000_01_01_00_04_05: REJECT POSITION CONTROL
2000_01_01_00_04_20: REJECT AUTO HOLD
2000_01_01_00_04_21: REJECT AUTO HOLD
2000_01_01_00_04_21: REJECT AUTO HOLD
2000_01_01_00_04_22: REJECT AUTO HOLD
2000_01_01_00_04_22: REJECT AUTO HOLD
2000_01_01_00_04_23: REJECT AUTO HOLD
2000_01_01_00_04_23: REJECT AUTO HOLD
2000_01_01_00_04_24: REJECT AUTO HOLD
2000_01_01_00_04_25: REJECT AUTO HOLD
2000_01_01_00_04_25: REJECT AUTO HOLD
2000_01_01_00_04_26: REJECT AUTO HOLD
2000_01_01_00_04_26: REJECT AUTO HOLD
2000_01_01_00_04_27: REJECT AUTO HOLD
2000_01_01_00_04_27: REJECT AUTO HOLD
2000_01_01_00_04_28: REJECT AUTO HOLD
2000_01_01_00_04_28: REJECT AUTO HOLD
2000_01_01_00_04_29: REJECT AUTO HOLD
2000_01_01_00_04_29: REJECT AUTO HOLD
2000_01_01_00_04_29: REJECT AUTO HOLD
2000_01_01_00_04_30: REJECT AUTO HOLD
2000_01_01_00_04_31: REJECT AUTO HOLD
2000_01_01_00_04_31: REJECT AUTO HOLD
2000_01_01_00_04_32: REJECT AUTO HOLD
2000_01_01_00_04_50: REJECT AUTO HOLD
2000_01_01_00_04_51: REJECT AUTO HOLD
2000_01_01_00_04_51: REJECT AUTO HOLD
2000_01_01_00_04_55: REJECT AUTO RTL
2000_01_01_00_04_56: REJECT AUTO HOLD
2000_01_01_00_05_00: REJECT AUTO HOLD
2000_01_01_00_05_00: REJECT AUTO HOLD
2000_01_01_00_05_01: REJECT AUTO HOLD
2000_01_01_00_05_02: REJECT AUTO HOLD
2000_01_01_00_05_03: REJECT AUTO HOLD
2000_01_01_00_05_05: REJECT POSITION CONTROL
2000_01_01_00_05_11: REJECT POSITION CONTROL
2000_01_01_00_00_04: DISARMED by RC
2000_01_01_00_00_04: DISARMED by RC
2000_01_01_00_00_46: REJECT POSITION CONTROL
2000_01_01_00_00_53: REJECT AUTO RTL
2000_01_01_00_00_56: REJECT AUTO HOLD
2000_01_01_00_01_07: REJECT AUTO HOLD
2000_01_01_00_01_16: MANUAL KILL SWITCH ENGAGED
2000_01_01_00_01_17: MANUAL KILL SWITCH OFF
2000_01_01_00_01_26: REJECT POSITION CONTROL
2000_01_01_00_01_29: REJECT AUTO HOLD
2000_01_01_00_01_30: REJECT AUTO RTL
2000_01_01_00_01_31: REJECT AUTO HOLD
2000_01_01_00_01_33: REJECT POSITION CONTROL
2000_01_01_00_01_34: REJECT AUTO RTL
2000_01_01_00_01_42: REJECT AUTO RTL
2000_01_01_00_01_51: REJECT AUTO HOLD
2000_01_01_00_02_01: REJECT OFFBOARD
2000_01_01_00_02_30: REJECT POSITION CONTROL
2000_01_01_00_02_43: REJECT POSITION CONTROL
2000_01_01_00_02_49: REJECT AUTO HOLD
2000_01_01_00_02_50: REJECT AUTO HOLD
2000_01_01_00_02_50: REJECT AUTO HOLD
2000_01_01_00_02_51: REJECT AUTO HOLD
2000_01_01_00_02_51: REJECT AUTO HOLD
2000_01_01_00_02_52: REJECT AUTO RTL
2000_01_01_00_02_54: REJECT AUTO HOLD
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_04_17: data link #0 lost
2000_01_01_00_00_04: REJECT AUTO RTL
2000_01_01_00_00_49: DISARMED by RC
2000_01_01_00_00_04: DISARMED by RC
2000_01_01_00_00_43: ARMED by RC
2000_01_01_00_00_44: TAKEOFF DETECTED
2000_01_01_00_00_44: [blackbox] /fs/microsd/log/sess001
2000_01_01_00_00_44: [blackbox] recording: log001.px4log
2000_01_01_00_00_51: DISARMED by RC
2000_01_01_00_00_52: TAKEOFF DETECTED
2000_01_01_00_00_52: [blackbox] stopped (0 drops)
2000_01_01_00_00_04: GPS fix lost
2016_05_26_19_00_43: REJECT AUTO RTL
2000_01_01_00_00_04: GPS fix lost
2016_05_31_16_44_40: GPS fix regained
2016_05_31_16_44_42: [inav] GPS signal found
2016_05_31_16_44_43: [inav] init ref: 47.1......, 7........, 473.3500
2016_05_31_16_45_27: ARMED by RC
2016_05_31_16_45_27: [blackbox] recording: 16_45_26.px4log
2016_05_31_16_45_31: TAKEOFF DETECTED
2016_05_31_16_45_37: DISARMED by RC
2016_05_31_16_45_37: TAKEOFF DETECTED
2016_05_31_16_45_38: [blackbox] stopped (1464 drops)
2016_05_31_16_45_41: ARMED by RC
2016_05_31_16_45_41: [blackbox] /fs/microsd/log/2016-05-31
2016_05_31_16_45_42: [blackbox] recording: 16_45_40.px4log
2016_05_31_16_45_43: TAKEOFF DETECTED
2016_05_31_16_45_47: DISARMED by RC
2016_05_31_16_45_47: TAKEOFF DETECTED
2016_05_31_16_45_48: [blackbox] stopped (0 drops)`

Strangely in the Log I can see the Actuators having PWM between 0 .. 900:
[image]

I had to unplug Power and reboot in order to fly....

julianoes, 2016-06-08:
Is this problem still persisting? If yes, can you please attach a log for investigation?

tops4u, 2016-12-06:
Problem was not seen after occuring 2-3 times. I assume fixed in current Versions.

closed 2016-12-06 by tops4u