PX4/PX4-Autopilot #4712: Gazebo sim: not arming

ChristophTobler, 2016-06-01:
when running "make posix_sitl_default gazebo_iris" arming does not work with neither mapping nor qgc gui. But it does work when typing "commander arm"

hsu, 2016-06-01:
this is not a solution, but rather just my debug work to try and discover why position lock fails. I made all topic namespaces absolute, see [this patch](https://github.com/PX4/sitl_gazebo/compare/master...hsu:master) and that seems to have enabled position lock, subsequently allowing simulated iris to arm/takeoff (`commander takeoff`). Maybe some one more familiar with the system can chime in.

If the problem is in fact caused by lack of absolute namespacing of topics, this suggests that the way gz transport deals with topic names has changed from gazebo 6 to gazebo 7.

LorenzMeier, 2016-06-01:
Sounds indeed like it. Is your patch an ok intermediate solution?

hsu, 2016-06-02:
If the missing slash (`/`) in front of topic name is indeed the problem, I'd like to add the proper namespaces - `/gazebo/[world name]/[model name]/imu` or possibly `/px4/[model name]/imu` for example. Let me do that first. Any suggestions?  Also tagging @nkoenig for comments about gz transport topic naming.

hsu, 2016-06-02:
feel free to commit my patch to get a working state. I'll make more PRs as needed.

ChristophTobler, 2017-03-15:
This seems fixed

closed 2017-03-15 by ChristophTobler