PX4/PX4-Autopilot #4714: LPE altitude drift

tumbili, 2016-06-01:
@jgoppert I just got a log from a user who is reporting considerable height drift during his flight.
Below you can see a plot which compares raw baro alt with gps alt and the estimated height.
Note that the baro / gps solutions are very close to the initial altitude at the end, however, the estimated height was suspect to a large drift.
 
[image]

jgoppert, 2016-06-02:
Something is very wrong with this log, there is no filter watch dog or estimator 0 states. I think we need to make sure these are working so I can diagnose the problem further, the problem may also be related to why this logging wasn't being recorded.

jgoppert, 2016-06-02:
Here is a recent log of LPE to compare, in this one we were flying outdoor with lidar and experienced a 50 m jump in GPS altitude from the sensor, which did cause our altitude to drift, but it recovered.

http://logs.uaventure.com/view/QdoUvByZrRHA2YLTxvZBTA

This flight was based on the 1.3.0rc1 release.

tumbili, 2016-06-02:
@jgoppert Yeah, I've been wondering why the filter states are not published, almost wanted to open an issue for that. So do you have any idea what the problem is?

jgoppert, 2016-06-02:
Is logging working for ekf2 on master? If it is, then I'm guessing there was a change to the pub interface and lpe needs to be updated or somehow a recent trivial change (see stack smashing and param src removal patches) I haven't flown yet with those patches.

jgoppert, 2016-06-02:
I will build and fly master to diagnose this on Friday night.

jgoppert, 2016-06-06:
I haven't been able to reproduce the altitude drift, see attached log and image:

[image]

http://logs.uaventure.com/view/9zJWm66UET9J4p4wRnq32W

I was able to reproduce the logging issue. I resolved this by commented out the isfinite checks on the estimator status. The numbers all appear finite after the initial estimator pub so this seems odd that it fixed it.

jgoppert, 2016-06-06:
I think this is might be due to a bad calibration of the pressure/gps offsets at boot, if you look at this plot, it looks like the global position altitude is converging to the GPS altitude.

[image]

The only way for me to say conclusively is another complete log of this event. I can't reproduce it on my end.

LorenzMeier, 2016-06-06:
@jgoppert what is the condition to initialize LPEs altitude estimate? Does GPS lock have to be good?

jgoppert, 2016-06-06:
@LorenzMeier 10 readings with eph < 3, n_sat >= 6

Maybe epv and lock type should also be a requirement?

https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/sensors/gps.cpp#L55

LorenzMeier, 2016-06-06:
You should certainly also factor those two in, yes.

jgoppert, 2016-06-06:
Log of gps stats over drift flight:
[image]

Log of gps stats over recent flight of master:
[image]

You can see that EPV was very high for this drift flight, so the offset theory may be reasonable, I'm adding that check as well as fix and creating a PR after I flight test.

jgoppert, 2016-06-07:
This isn't even running LPE, SYS_MC_EST_GROUP = 0.0

This is an INAV bug. You can see it even in the log parameters. I just noticed it because Ilia gave me all the files off of his sdcard and I noticed that they were never running lpe

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  106873 57.016     0/    0   0 (  0)  READY 
   1 hpwork                       4714  2.944   844/ 1592 192 (192)  w:sig 
   2 lpwork                        775  0.294   572/ 1592  50 ( 50)  READY 
   3 init                         1667  0.000  1324/ 2496 100 (100)  w:sem 
 243 sdlog2_writer                  33  3.238  1140/ 2040  95 ( 95)  w:sem 
  90 gps                           350  0.196   804/ 1192 220 (220)  w:sem 
  92 dataman                        29  0.000   660/ 1192  90 ( 90)  w:sem 
 118 sensors                      4994  2.845  1724/ 1992 250 (250)  w:sem 
 120 commander                    3234  1.766  3220/ 3592 140 (140)  READY 
 121 commander_low_prio            499  0.000   788/ 2992  50 ( 50)  w:sem 
 127 px4io                        5155  3.042  1060/ 1392 240 (240)  w:sem 
 137 mavlink_if0                  1485  0.686  2356/ 2792 100 (100)  READY 
 138 mavlink_rcv_if0                72  0.000  1192/ 2096 175 (175)  w:sem 
 145 mavlink_if1                  1182  0.686  2356/ 2792 100 (100)  READY 
 146 mavlink_rcv_if1                12  0.000   988/ 2096 175 (175)  w:sem 
 180 sdlog2                       1858  3.140  2996/ 3392 177 (177)  RUN   
 215 attitude_estimator_q         9158  5.495  1996/ 2496 250 (250)  w:sem 
 **221 position_estimator_inav      4192  3.434  4828/ 5296 250 (250)  w:sem** 
 230 mc_att_control               5205  3.238  1228/ 1496 250 (250)  w:sem 
 235 mc_pos_control               1681  1.079  1180/ 1896 250 (250)  w:sem 
 239 navigator                     270  0.686   940/ 1296 105 (105)  w:sem

closed 2016-06-07 by jgoppert