PX4/PX4-Autopilot #4723: SITL Gazebo broken on master

sanderux, 2016-06-02:
Sitl gazebo no longer functions on current master.
Jmavsim does work.

Tested: 
- make posix_sitl_default gazebo_standard_vtol
- make posix_sitl_default gazebo

Both do start gazebo but no connection and continues timeouts

On older branches it does work.
ubuntu 14.04

```
WARN  log buffer size: 8192 bytes
INFO  simulator is slow. Delay added: 195761 us
INFO  [blackbox] rootfs/fs/microsd/log/2016-06-02
WARN  sdlog2: failed setting sched params
INFO  [blackbox] recording: 21_16_39.px4log
INFO  simulator is slow. Delay added: 193338 us
INFO  simulator is slow. Delay added: 193484 us
pxh> INFO  simulator is slow. Delay added: 195032 us
INFO  simulator is slow. Delay added: 196528 us
INFO  simulator is slow. Delay added: 195460 us
INFO  simulator is slow. Delay added: 192877 us
INFO  simulator is slow. Delay added: 195833 us
INFO  simulator is slow. Delay added: 196632 us
INFO  simulator is slow. Delay added: 195057 us
INFO  simulator is slow. Delay added: 197594 us
```

AndreasAntener, 2016-06-03:
I have this issue currently on Mac but only with `gazebo_standard_vtol`

julianoes, 2016-06-03:
Can you try gazebo 7?

Does the gazebo viewer close some seconds after startup?

sanderux, 2016-06-03:
I prefer not to change a working setup as it was quite hard to get it running and stable. Viewer did not close but the props were not spinning, it seemed to hang

julianoes, 2016-06-03:
Hm, that's different from what I see. For me the viewer closes.

sanderux, 2016-06-03:
I will try this weekend if i have time

sanderux, 2016-06-03:
@julianoes i tried updating to gazebo7, no effect

sanderux, 2016-06-03:
@julianoes bisect came up with:

```
bdf064fd8f8d5b7d3ef4a937f2dfc9e4e7b26551 is the first bad commit
commit bdf064fd8f8d5b7d3ef4a937f2dfc9e4e7b26551
Author: Lorenz Meier <lm@inf.ethz.ch>
Date:   Thu Jun 2 06:45:32 2016 +0200

    Update SITL Gazebo plugin
```

Makes sense :)

/cc @LorenzMeier

julianoes, 2016-06-05:
@sanderux thanks for bisecting. I believe you that this commit breaks it for you but I don't have an explanation why because it doesn't for me. I tried both gazebo 6 and 7.

sanderux, 2016-06-05:
What OS did you try? I'm on ubuntu 14.04

julianoes, 2016-06-05:
I'm on Ubuntu 15.10.

sanderux, 2016-06-05:
Does gazebo_standard_vtol work for you?

commit ed0aad61 (2016-06-05): Updated SITL gazebo, fixes #4723
closed 2016-06-05 by LorenzMeier