PX4/PX4-Autopilot #4728: frsky_telemetry not reporting on D4R-II and Taranis Plus

yashm, 2016-06-03:
1. frsky_telemetry not sending any data to Serial4 on Pixhawk was fixed by changing
[Line 217 in frsky_telemetery.c](https://github.com/PX4/Firmware/blob/master/src/drivers/frsky_telemetry/frsky_telemetry.c#L217) from
`} else if (status > 0) {`
to
`} else if (status >= 0) {`

2. After making the above change, serial data can be read off of Serial4 on a serial monitor through an FTDI adapter.

3. Adding `warnx("Received %d bytes", status);` on [Line 211 in frsky_telemetery.c](https://github.com/PX4/Firmware/blob/master/src/drivers/frsky_telemetry/frsky_telemetry.c#L211) yields the following output in the NuttShell
`frsky_telemetry: Received 0 bytes`

Thanks to @thedevleon for his help.

kd0aij, 2016-06-04:
This indicates that the Pixhawk isn't seeing serial data coming from the D4R receiver. Probably a cable problem. Master is working properly on Pixracer using the standard port /dev/ttyS6

mhkabir, 2016-06-04:
@yashm, 1 isn't a fix. You're just forcing it to send data, overriding the auto detection.

mhkabir, 2017-04-03:
Please reopen if still valid.

closed 2017-04-03 by mhkabir