PX4/PX4-Autopilot #4736: Feature request for easier regulatory permitting of PX4

jcooper21, 2016-06-04:
It has come to my attention for the sake of commercial operation and permitting a more advanced geofence feature can make permitting PX4 much easier in North America. 

A prevent breach option when setting up the geofence would be ideal. The 3DR Solo and DJI copters all have this feature.  It would be great if PX4 had this for copters and planes.

dagar, 2016-06-07:
Can you expand on how this would be different than the current geofence? Do you want the current functionality but the fence to be more intelligent so it takes control before breach instead of after?
Or are you talking about adding keep out areas?

jcooper21, 2016-06-07:
If the geofence is set to 100m if the plane hits 100 to 101m then it pushes the plane back down to 100m rather than just saying it breached by X amount or go into loiter. 

For distance it prevents a breach if it breaches it turns it back into the geofence area and contains it.

dagar, 2016-06-07:
So it sounds like the geofence max vertical distance should also be an altitude cap. That would apply in any mode that controls altitude, but you'd still get the current behaviour if you exceeded in manual/stabilized. 

Isn't the distance behaviour that you described what happens now if you've set the "geofence violation action" to RTL? One possible improvement here is making it more intelligent so it takes over in time to stop you from breaching the fence, but that's a bit harder to implement with the current polygon geofence.

jcooper21, 2016-06-07:
Yes I think you have nailed it there with the altitude cap.

closed 2018-02-21 by PX4BuildBot