PX4/PX4-Autopilot #4740: Can't write top in NSH

CarlOlsson, 2016-06-06:
I just flashed [master](f9df60919e4722edade53b066d25248af9d73fd6) to two different pixracer boards and I can't write top in nsh. I only get "2J" back.
Is anyone else experiencing this?

CarlOlsson, 2016-06-06:
It's on the wingwing fixed wing airplane and standard ekf running

devbharat, 2016-06-06:
I just checked the available list of nuttx commands, top isn't there anymore. This is the update/subtree branch 

[image]

CarlOlsson, 2016-06-06:
On standard master i get this output, which included the top command as a built in app

nsh> \0x1b[Khelp
help usage:  help [-v] [<cmd>]

  [           dd          hexdump     mkfifo      pwd         umount      
  ?           df          kill        mkrd        rm          unset       
  cat         echo        losetup     mh          rmdir       usleep      
  cd          exec        ls          mount       set         xd          
  cp          exit        mb          mv          sh          
  cmp         free        mkdir       mw          sleep       
  date        help        mkfatfs     ps          test        

Builtin Apps:
  adc
  tone_alarm
  fmu
  rgbled
  mpu6000
  mpu9250
  hmc5883
  ms5611
  mb12xx
  srf02
  sf0x
  ll40ls
  trone
  gps
  pwm_out_sim
  hott_telemetry
  hott_sensors
  blinkm
  ets_airspeed
  meas_airspeed
  frsky_telemetry
  sensors
  mkblctrl
  px4flow
  oreoled
  gimbal
  pwm_input
  camera_trigger
  bst
  snapdragon_rc_pwm
  lis3mdl
  bmp280
  bma180
  bmi160
  bl_update
  mixer
  param
  perf
  pwm
  esc_calib
  reboot
  listener
  top
  config
  nshterm
  mtd
  dumpfile
  ver
  sd_bench
  tests
  commander
  load_mon
  navigator
  mavlink
  gpio_led
  uavcan
  land_detector
  attitude_estimator_q
  ekf_att_pos_estimator
  position_estimator_inav
  ekf2
  local_position_estimator
  fw_pos_control_l1
  fw_att_control
  mc_att_control
  mc_pos_control
  vtol_att_control
  sdlog2
  logger
  uorb
  dataman
  bottle_drop
  rover_steering_control
  px4_simple_app
  serdis
  sercon

LorenzMeier, 2016-06-06:
Do you connect with a VT100-enabled client? Because it looks you might not. Try to run a fresh instance of screen.

CarlOlsson, 2016-06-06:
Yes it works with screen, thanks. Everything is as it should and closing here?

LorenzMeier, 2016-06-06:
Yes - you just picked an incompatible terminal.

closed 2016-06-06 by LorenzMeier