PX4/PX4-Autopilot #4742: Where does our memory go?

bkueng, 2016-06-06:
Since memory is scarce, I thought a different analysis gives some more insights, where the memory goes. We already analyzed stack sizes and static code size in general, so I did a heap analysis.

The result is this [nice graph](https://github.com/PX4/Firmware/files/300334/heap.pdf). I used `gperftools` under linux SITL. The numbers are all zero, because they are in MB, just look at the percentages. They show the live memory (of the node and the subtree), meaning the memory that was still in use at the end.
Not surprisingly, the logger needs a lot, but there are three others that stick out:
- orb
- ekf2
- param_set_internal

I'm most familiar with orb, so let's zoom in on that. `uORB::DeviceNode::open` allocates the `SubscriberData`:
```
	struct SubscriberData {
		unsigned  generation; /**< last generation the subscriber has seen */
		unsigned  update_interval; /**< if nonzero minimum interval between updates */
		uint64_t last_update; /**< time at which the last update was provided, used when update_interval is nonzero */
		struct hrt_call update_call;  /**< deferred wakeup call if update_period is nonzero */
		void    *poll_priv; /**< saved copy of fds->f_priv while poll is active */
		bool    update_reported; /**< true if we have reported the update via poll/check */
		int   priority; /**< priority of publisher */
	};
```
This is 64 Bytes and I counted 180 instances, so 11kb memory. Without much effort & overhead we can cut this down to about half: `update_call` is large and only used if `orb_set_interval` is used, which is not often the case. We can make this a pointer and combine it with `update_interval`. `update_reported` is only 1 bit and can be merged into `priority`. `poll_priv` is some leftover that is not referenced.

What the graph does not show is the stack sizes and NuttX overhead and differences between SITL & real hardware.
I used the following commands to create the graph:
- run jmavsim: `cd Tools/jMAVSim/out/production && java -Djava.ext.dirs= -jar jmavsim_run.jar -udp 127.0.0.1:14560`
- then:
```
  cd build_posix_sitl_default/src/firmware/posix
  export HEAPPROFILE=/tmp/heapprofile.hprof
  env LD_PRELOAD=/lib64/libtcmalloc.so ./mainapp ../../../../posix-configs/SITL/init/rcS_jmavsim_iris
  pprof --pdf mainapp /tmp/heapprofile.hprof.0001.heap
```

LorenzMeier, 2016-06-06:
Nice analysis, impressive!

LorenzMeier, 2016-06-06:
@bkueng The param system should not be that heavy on NuttX - but we should check. We need to make  sure param names never get copied but stay const.

LorenzMeier, 2016-06-06:
Just did the math. Its certainly not strings. Need to see if we can make the param vans more lightweight.

bkueng, 2016-06-08:
Reduction of `SubscriberData` is here: https://github.com/PX4/Firmware/pull/4771

This is the [updated graph](https://github.com/PX4/Firmware/files/304483/heap_new.pdf). In several runs, `param` does not stick out anymore. Not sure what happened the first time with `params`, but I think it's fine.

bkueng, 2017-04-07:
Closing as we have instructions on the dev-guide for this.

closed 2017-04-07 by bkueng