PX4/PX4-Autopilot #4751: Battery Voltage Gain Scaling for Motor Outputs

jgoppert, 2016-06-07:
Lipo batteries typically have a 10-20% change in voltage over the course of a flight. Because of this, it appears the motor gains are also changing by ~20% during a flight. On some frames, this makes it very hard to find gains that fly well during the entire duration. You can also see this in the throttle output of logs:

http://logs.uaventure.com/view/4RFQF4wAGZjLDe9koXhJ5m#Altitude_PLOT 

Notice the mean throttle increases by about 10% over the course of this flight.

I believe we would get better performance if we scale the motor outputs by the inverse of the battery voltage percentage, effectively normalizing the output. For instance, if the max battery voltage is 25 volts, and the current battery is reading 22 volts, then the gain to the motor should be:

```
kP_adjusted = kP * (25/22)
```

This is just a rough guess, because the trend looks very linear to me.

jgoppert, 2016-06-07:
[image]

[image]

jgoppert, 2016-06-07:
Here are the commands I used to generate the above plots: 

```python
scaled = (data_jmg.ATSP_ThrustSP * (data_jmg.BATT_VFilt/10.95))
orig = data_jmg.ATSP_ThrustSP
pl.figure(figsize=(20,5))
scaled.ewm(span=1000).mean().plot(label='scaled')
orig.ewm(span=1000).mean().plot(label='orig')
pl.legend(loc='best')
pl.axis([20,190,0.6,0.8])
pl.grid()
pl.title('motor voltage scaling')
pl.figure(figsize=(20,5))
(data_jmg.BATT_VFilt/10.95).plot()
pl.title('scale factor')
```

You scan see that the bottom curve is much more level that the top.

jcooper21, 2016-06-07:
I would like to add lithium ion have an even larger voltage range than Lipos from 4.2 volts/cell down to 2.5 volts/cell so it had an even larger effect with those batteries.

jgoppert, 2016-06-08:
@LorenzMeier thoughts on this? Should be easy to implement.

LorenzMeier, 2016-06-08:
Sounds like a good idea, yes. Just required enough test time.

commit a634c145 (2016-11-19): Scale throttle outputs by battery level. Fixes #4751
closed 2016-11-19 by LorenzMeier