PX4/PX4-Autopilot #4753: Pre-programmed flight primitives

LorenzMeier, 2016-06-07:
@tumbili @bkueng We need support for pre-programmed flight primitives like for example flying backwards and up for a zoom-out motion as offered by today's common consumer systems. These primitives should be available in missions, but we should also be able to trigger them from the remote and modify them with stick input.

tubeme, 2016-06-07:
Very good idea! I was thinking about dedicated set of Film and TV shooting modes specially tailored to  the movie and tv shooting pros. Like for example for your idea I thought that in this zoom mode if I could control the whole backward and up movement with just the pitch.. Like if I pitch down the quad should go back and upward. If I pitch up than the quad goes forward and down. Another useful primitive could be making a cylinder spin around an object. I go to the center of the object. turn the spin mode on, the quad locks the center point. Then I move backward until I'm far away from the center spot with the radius i want. And then when I roll the quad it starts making this circle around the center spot. Then with throttle I control the height. This way I can fully control the quad in the spiral. Or course I can do this with roll and yaw and throttle, but is it hard sometimes to be careful about the shot framing and at the same time to pilot the perfect spiral. There are a lot of ideas. I was shooting a lot of productions for the last 15 years and now I'm shooting a lot for TV and ads with my MCs so this kind of additions will make PX4 invaluable in the production set. Because I haven't found yet a controller well thought from production and movie shooting stand point.

LorenzMeier, 2016-06-29:
@tumbili Can you please connect to @dogmaphobic for the QGC side support?

LorenzMeier, 2016-06-29:
Only doing circle for now.

closed 2016-06-29 by LorenzMeier

dogmaphobic, 2016-06-29:
For this to work you would need coordination with the gimbal so the camera is pointing to the right place as the aircraft moves about.