PX4/PX4-Autopilot #4754: jMAVSim installation fails

Namorus, 2016-06-07:
Hi everyone
I've got a problem with the installation of jMAVSim. 
I followed the guide provided: https://pixhawk.org/dev/hil/jmavsim
Unfortunately I'm not able to build jMAVSim.

maverick@Mavericks-MacBook-Pro:~/jMAVSim$ **git clone https://github.com/PX4/jMAVSim.git**
Cloning into 'jMAVSim'...
remote: Counting objects: 2248, done.
remote: Total 2248 (delta 5), reused 5 (delta 5), pack-reused 2242
Receiving objects: 100% (2248/2248), 25.91 MiB | 3.06 MiB/s, done.
Resolving deltas: 100% (1130/1130), done.
Checking connectivity... done.
maverick@Mavericks-MacBook-Pro:~/jMAVSim$ **brew install ant**
Warning: ant-1.9.7 already installed
maverick@Mavericks-MacBook-Pro:~/jMAVSim$ **cd jMAVSim/**
maverick@Mavericks-MacBook-Pro:~/jMAVSim/jMAVSim$ **git submodule init**
Submodule 'jMAVlib' (https://github.com/DrTon/jMAVlib.git) registered for path 'jMAVlib'
maverick@Mavericks-MacBook-Pro:~/jMAVSim/jMAVSim$ **git submodule update**
Cloning into '/Users/maverick/jMAVSim/jMAVSim/jMAVlib'...
remote: Counting objects: 576, done.
remote: Total 576 (delta 0), reused 0 (delta 0), pack-reused 576
Receiving objects: 100% (576/576), 95.31 KiB | 0 bytes/s, done.
Resolving deltas: 100% (221/221), done.
Checking connectivity... done.
Submodule path 'jMAVlib': checked out '53f51ed9969472c82119ff2edde9c62b7b70eb94'
maverick@Mavericks-MacBook-Pro:~/jMAVSim/jMAVSim$ **ant**
Buildfile: /Users/maverick/jMAVSim/jMAVSim/build.xml

make_dirs:
    [mkdir] Created dir: /Users/maverick/jMAVSim/jMAVSim/out/production/jMAVSim

compile:
    [javac] Compiling 64 source files to /Users/maverick/jMAVSim/jMAVSim/out/production/jMAVSim
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: <identifier> expected
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                             ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: '{' expected
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                               ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: illegal start of expression
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                                                         ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: ';' expected
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                                                                                ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: not a statement
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                                                                                 ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: ';' expected
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                                                                                  ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:86: illegal start of expression
    [javac]     public BranchGroup getBranchGroup() {
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:86: ';' expected
    [javac]     public BranchGroup getBranchGroup() {
    [javac]                                      ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:90: illegal start of expression
    [javac]     public void updateBranchGroup() {
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:90: illegal start of expression
    [javac]     public void updateBranchGroup() {
    [javac]            ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:90: ';' expected
    [javac]     public void updateBranchGroup() {
    [javac]                                  ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:96: illegal start of expression
    [javac]     public void setIgnoreGravity(boolean ignoreGravity) {
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:96: illegal start of expression
[... 244 lines cut ...]
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/Simulator.java:434: illegal start of expression
    [javac]     public final static String MAG_STRING = "-automag";
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/Simulator.java:435: illegal start of expression
    [javac]     public final static String REP_STRING = "-rep";
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/Simulator.java:436: illegal start of expression
    [javac]     public final static String GUI_AA_STRING = "[-no]-aa";
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/Simulator.java:437: illegal start of expression
    [javac]     public final static String GIMBAL_STRING = "[-no]-gimbal";
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/Simulator.java:438: illegal start of expression
    [javac]     public final static String GUI_MAX_STRING = "-max";
    [javac]     ^
    [javac] /Users/maverick/jMAVSim/jMAVSim/src/me/drton/jmavsim/Simulator.java:439: illegal start of expression
    [javac]     public final static String GUI_VIEW_STRING = "-view (fpv|grnd|gmbl)";
    [javac]     ^
    [javac] 100 errors

BUILD FAILED
/Users/maverick/jMAVSim/jMAVSim/build.xml:25: Compile failed; see the compiler error output for details.

Total time: 1 second

Namorus, 2016-06-07:
I gave it another try with original git repository: https://github.com/DrTon/jMAVSim
This time, the installation seems fine. 

maverick@Mavericks-MacBook-Pro:~/jMAVSim$ **git clone https://github.com/DrTon/jMAVSim**
Cloning into 'jMAVSim'...
remote: Counting objects: 1448, done.
remote: Total 1448 (delta 0), reused 0 (delta 0), pack-reused 1448
Receiving objects: 100% (1448/1448), 13.15 MiB | 2.35 MiB/s, done.
Resolving deltas: 100% (779/779), done.
Checking connectivity... done.
maverick@Mavericks-MacBook-Pro:~/jMAVSim$ **git submodule init**
fatal: Not a git repository (or any of the parent directories): .git
maverick@Mavericks-MacBook-Pro:~/jMAVSim$ **cd jMAVSim/**
maverick@Mavericks-MacBook-Pro:~/jMAVSim/jMAVSim$ **git submodule init**
Submodule 'jMAVlib' (https://github.com/DrTon/jMAVlib.git) registered for path 'jMAVlib'
maverick@Mavericks-MacBook-Pro:~/jMAVSim/jMAVSim$ **git submodule update**
Cloning into '/Users/maverick/jMAVSim/jMAVSim/jMAVlib'...
remote: Counting objects: 576, done.
remote: Total 576 (delta 0), reused 0 (delta 0), pack-reused 576
Receiving objects: 100% (576/576), 95.31 KiB | 0 bytes/s, done.
Resolving deltas: 100% (221/221), done.
Checking connectivity... done.
Submodule path 'jMAVlib': checked out '53f51ed9969472c82119ff2edde9c62b7b70eb94'
maverick@Mavericks-MacBook-Pro:~/jMAVSim/jMAVSim$ **ant**
Buildfile: /Users/maverick/jMAVSim/jMAVSim/build.xml

make_dirs:
    [mkdir] Created dir: /Users/maverick/jMAVSim/jMAVSim/out/production/jMAVSim

compile:
    [javac] Compiling 58 source files to /Users/maverick/jMAVSim/jMAVSim/out/production/jMAVSim

jmavsim:
      [jar] Building jar: /Users/maverick/jMAVSim/jMAVSim/out/production/jmavsim.jar

all:

BUILD SUCCESSFUL
Total time: 3 seconds

if I change now to my firmware folder and follow the guide: http://dev.px4.io/simulation-sitl.html
Here the build fails again. Am I doing something wrong in principle or do I have to insert a fix somewhere?

I get this message: 

maverick@Mavericks-MacBook-Pro:~$ **cd src/Firmware/**
maverick@Mavericks-MacBook-Pro:~/src/Firmware$ **make posix_sitl_default jmavsim**
PX4 CONFIG: posix_sitl_default
[0/1] cd /Users/maverick/src/Firmware && Tools/sitl...sers/maverick/src/Firmware/build_posix_sitl_defaul
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: jmavsim
model: none
build_path: /Users/maverick/src/Firmware/build_posix_sitl_default
empty model, setting iris as default
Buildfile: /Users/maverick/src/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:
    [javac] Compiling 64 source files to /Users/maverick/src/Firmware/Tools/jMAVSim/out/production/jMAVSim
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: <identifier> expected
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                             ^
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: '{' expected
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
    [javac]                               ^
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/KinematicObject.java:79: illegal start of expression
    [javac]         } catch (IOException | IncorrectFormatException | ParsingErrorException e) {
[... 277 lines cut ...]
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/Simulator.java:435: illegal start of expression
    [javac]     public final static String REP_STRING = "-rep";
    [javac]     ^
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/Simulator.java:436: illegal start of expression
    [javac]     public final static String GUI_AA_STRING = "[-no]-aa";
    [javac]     ^
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/Simulator.java:437: illegal start of expression
    [javac]     public final static String GIMBAL_STRING = "[-no]-gimbal";
    [javac]     ^
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/Simulator.java:438: illegal start of expression
    [javac]     public final static String GUI_MAX_STRING = "-max";
    [javac]     ^
    [javac] /Users/maverick/src/Firmware/Tools/jMAVSim/src/me/drton/jmavsim/Simulator.java:439: illegal start of expression
    [javac]     public final static String GUI_VIEW_STRING = "-view (fpv|grnd|gmbl)";
    [javac]     ^
    [javac] 100 errors

BUILD FAILED
/Users/maverick/src/Firmware/Tools/jMAVSim/build.xml:25: Compile failed; see the compiler error output for details.

Total time: 1 second
FAILED: src/firmware/posix/CMakeFiles/jmavsim 
cd /Users/maverick/src/Firmware && Tools/sitl_run.sh posix-configs/SITL/init/rcS none jmavsim none /Users/maverick/src/Firmware/build_posix_sitl_default
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

MHageH, 2016-06-07:
Following the same procedure on [jmavsim tutorial] (https://pixhawk.org/dev/hil/jmavsim), the compilation runs successfully on Linux Debian x64
```
$ git clone https://github.com/PX4/jMAVSim.git
$ cd  jMAVSIM
$ git submodule init
$ git submodule update
$ ant

Buildfile: /home/dl0/src/TEST/jMAVSIM/build.xml

make_dirs:
    [mkdir] Created dir: /home/dl0/src/TEST/jMAVSIM/out/production/jMAVSim

compile:
    [javac] Compiling 64 source files to /home/dl0/src/TEST/jMAVSIM/out/production/jMAVSim

jmavsim:
      [jar] Building jar: /home/dl0/src/TEST/jMAVSIM/out/production/jmavsim.jar

all:

BUILD SUCCESSFUL
Total time: 3 seconds
```
Following the original PX4 simulation procedure : [SITL](http://dev.px4.io/simulation-sitl.html), the firmware runs in SITL mode too (compilation successful).
Did you follow the [Mac OS guide](http://dev.px4.io/starting-installing-mac.html) correctly?

Namorus, 2016-06-07:
I followed the Mac OS guide again and it seems that everything is installed correctly:
(by the way: I'm running Mac OSX El Capitan)

maverick@Mavericks-MacBook-Pro:~$ **brew tap PX4/homebrew-px4**
maverick@Mavericks-MacBook-Pro:~$ **brew tap osrf/simulation**
maverick@Mavericks-MacBook-Pro:~$ **brew update**
Already up-to-date.
maverick@Mavericks-MacBook-Pro:~$ **brew install git bash-completion genromfs kconfig-frontends gcc-arm-none-eabi**
Warning: git-2.8.3 already installed
Warning: bash-completion-1.3_1 already installed
Warning: px4/px4/genromfs-0.5.2 already installed
Warning: px4/px4/kconfig-frontends-3.7.0.0 already installed
Warning: px4/px4/gcc-arm-none-eabi-20140805 already installed
maverick@Mavericks-MacBook-Pro:~$ brew install astyle cmake ninja
Warning: astyle-2.05.1 already installed
Warning: cmake-3.5.2 already installed
Warning: ninja-1.7.1 already installed
maverick@Mavericks-MacBook-Pro:~$ **brew install ant graphviz sdformat3 protobuf eigen opencv**
Warning: ant-1.9.7 already installed
Warning: graphviz-2.38.0 already installed
Warning: osrf/simulation/sdformat3-3.7.0 already installed
Warning: protobuf-2.6.1 already installed
Warning: eigen-3.2.8 already installed
Warning: homebrew/science/opencv-2.4.13 already installed
maverick@Mavericks-MacBook-Pro:~$ **sudo easy_install pip**
Password:
Searching for pip
Best match: pip 8.1.2
Processing pip-8.1.2-py2.7.egg
pip 8.1.2 is already the active version in easy-install.pth
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Using /Library/Python/2.7/site-packages/pip-8.1.2-py2.7.egg
Processing dependencies for pip
Finished processing dependencies for pip
maverick@Mavericks-MacBook-Pro:~$ **sudo pip install pyserial empy**
The directory '/Users/maverick/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/maverick/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied (use --upgrade to upgrade): pyserial in /Library/Python/2.7/site-packages
Requirement already satisfied (use --upgrade to upgrade): empy in /Library/Python/2.7/site-packages

Namorus, 2016-06-07:
Wrong java version was the problem. I had Java SE Runtime Environment 8  installed, which was obviously the **wrong one**:
http://www.oracle.com/technetwork/java/javase/downloads/jre8-downloads-2133155.html
**Java SE Development Kit 8** solved the problem:
http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html

closed 2016-06-07 by Namorus