PX4/PX4-Autopilot #4759: Snapdragon: sometimes QURT build takes a very long time

julianoes, 2016-06-07:
Recently I noticed that "sometimes" a clean build for QURT takes an awful long time.
This has happened both with Qualcomm SDK 2.0 and SDK 3.0.

The moment where it happens is `HERE`:

```
[ 98%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/param/param_shmem.c.o
[ 99%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/print_load_posix.c.o
[ 99%] Linking CXX static library libmodules__systemlib.a
[ 99%] Built target modules__systemlib
[ 99%] Linking CXX static library liblib__ecl.a
[ 99%] Built target lib__ecl
[ 99%] Linking CXX static library libmodules__commander.a
[ 99%] Built target modules__commander
HERE
[ 99%] Linking CXX static library libmodules__ekf_att_pos_estimator.a
[ 99%] Built target modules__ekf_att_pos_estimator
Scanning dependencies of target mainapp
[ 99%] Linking CXX shared library libmainapp.so
[ 99%] Built target mainapp
Scanning dependencies of target px4muorb_skel
[100%] Building C object src/firmware/qurt/CMakeFiles/px4muorb_skel.dir/px4muorb_skel.c.o
[100%] Linking CXX shared module libpx4muorb_skel.so
[100%] Built target px4muorb_skel
```

@mcharleb any idea?

mhkabir, 2016-06-08:
I believe that this started happening after we enabled more compiler optimizations. 
Disabling build of older att-pos EKF was a workaround that worked for me.

mcharleb, 2016-09-20:
The hexagon toolchain seems to spend a huge amount of time trying to optimize the att-pos EKF code.

julianoes, 2016-09-27:
> @mhkabir I believe that this started happening after we enabled more compiler optimizations. 
Disabling build of older att-pos EKF was a workaround that worked for me.

I tend to agree, we should make the optimization flags platform dependant.

julianoes, 2016-09-28:
Addressed in #5554.

julianoes, 2016-10-22:
Well, ecl just takes it's time, especially with compile flags, not much we can do.

closed 2016-10-22 by julianoes