PX4/PX4-Autopilot #4760: Can’t connect to SITL using Dronekit – mode is 0

okalachev, 2016-06-07:
Checked out v1.3.1 version of the Firmware. Built and ran SITL with the command:
````
make posix_sitl_default jmavsim
```

JMavSim ran successfully. Trying to connect to the copter like this:
```python
vehicle = connect('udp:127.0.0.1:14556', wait_ready=False)
```

 Getting an exception:
```
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
```

Also, APM Planner shows, that the vehicle is in UNKNOWN mode. This can’t be changed neither with APM Planner/Mission Planner, nor with Mavproxy console.

Anyone can help?

LorenzMeier, 2016-06-07:
Dronekit support is work in progress. You have to use QGroundControl, not APM Planner.

Can you file an issue in Dronekit for the mode announcement?

And please look at http://dev.px4.io for a Dronekit example.

closed 2016-06-07 by LorenzMeier

okalachev, 2016-06-07:
I’m sorry, I don’t completely get you answer.

Is mode «0» a correct mode, that is, for some reason, isn’t defined in DroneKit? Also, I can’t change it from DroneKit, to «GUIDED» for example, is that a DroneKit issue?

okalachev, 2016-06-07:
Also, not I tried 14550 UDP port, and got another exception:
```
mode 65536 not available on mavlink definition
```

LorenzMeier, 2016-06-07:
It would really help if you would consider reading the documentation I referenced earlier:
http://dev.px4.io/dronekit-example.html

okalachev, 2016-06-07:
OK, thanks, I read it. So, it’s impossible to switch mode with DroneKit’s vehicle.mode property? But I was able to do that on a real PX4. 
Anyways, the example file on the manual doesn’t connect to the simulation copter either, it hangs with the same exception:
```
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
```

LorenzMeier, 2016-06-07:
That's because you didn't follow the instructions - did you see that they require you to install the latest Dronekit from source?

okalachev, 2016-06-07:
Using the latest DroneKit from the sources. Got the same exception.

okalachev, 2016-06-07:
Also, I would appreciate very much, it you made it more clear for me. Now, I’m trying to make an autonomous flight on a drone using DroneKit.

You said, that supporting DroneKit is in progress, so does the PX4 support following DroneKit’s features?

- simple_takeoff? (worked for me on a real drone)
- simple_goto? (didn’t try)
- controlling the speeds by SET_POSITION_TARGET_LOCAL_NED MAVLink message? (didn’t worked for me)
- controlling the yaw by CONDITION_YAW message?
- RTL mode?

I’ve been messing up with this for a couple of weeks, and still got no results :(

jukindle, 2018-09-17:
Still the same issue here...

okalachev, 2018-09-17:
For communication with PX4 I recommend using mavros or pymavlink.