PX4/PX4-Autopilot #4764: VTOL: Nasty Yaw problem after back trans. in ALT and POS

tubeme, 2016-06-07:
Please help me solve this mystery.

After a long way with your help we've managed to make almost perfect transitions in Stabalized mode. But recently started to do Altitude and Position mode transitions and now we have e different set of problems.

The front transition is rock solid in all modes, but we have a very strange problem with the yaw handling after we make back transition in the assisted modes. The yaw excites itself somehow, without any input from the RC. Even sometimes starting to yaw by itself 180 degrees and than back..

You could clearly see this in this graphic.This time it did it just in the transition::

[image]

And then after a couple of seconds in hovering state in Position it did this:

[image]

I really try to find the problem for the past week and I'm clueless...

Here is the whole log.

http://logs.uaventure.com/view/w4ETHE2TY5qFMfrqrzrLoS

And another flight with the same problem from the same day.

http://logs.uaventure.com/view/zkMN6WPaKeC6MZiMupFi86

Any help will be appreciated.

tumbili, 2016-06-08:
@tubeme Hi Vasil, I had a look at your log and I think you have brought yourself in a nasty situation.
I was having a look at the situation where the vehicle was turning around several times in yaw although the yaw setpoint was fixed at roughly 0 degrees. Due to the relatively big inertia of the plane and the weak yaw authority of the motors the controller got into the situation where it couldn't reduce the yaw speed at the desired heading. Instead it overshot by more than 180 degrees and as a result tried to approach the desired heading from the opposite direction again. This went on like this for a few times. Eventually, when you switched to manual the controller managed to slow down the vehicle and keep the heading at the desired value. It is noticeable that this only happened when you switched back to manual. An explanation for this can be that the mixer was reducing yaw control in position control mode in order to guarantee sufficient roll and pitch authority. If you have a look at ATTC.Roll and ATTC.Pitch then you see that those are much higher before you switched to manual.
Have you tried out the VTOL weathervane strategy?
[image]

tubeme, 2016-06-08:
@tumbili Hi Roman, thanks for the look and here is more info because I understand exactly what you are saying and in this situation I have no solution so the dev team is my only hope.

> Have you tried out the VTOL weathervane strategy?

This is not applicable to our VTOL, because it is flying wing design. We get YAW control only after we make the transition.

> Due to the relatively big inertia of the plane and the weak yaw authority of the motors the controller got into the situation where it couldn't reduce the yaw speed at the desired heading. 

This is exactly what is happening in the transition. We have a huge inertial moments when transitioning from FW and the yaw authority is very weak. We very easily can recreate the same problem in Manual btw. It is done by gliding the plane on 20%-30% throttle and then flip the transition. In this situation due to very low throttle, the quad could hot gain control over the machine so the machine starts sliding forward and at some point does this 90-180 degree yaw swings left and right. After this even if we push the throttle to max the machine often gets the leaf effect falling down, and PX4 is not able to fully recover if there is no 30-40 meters of height and it just crashes. 

I cranked the YAW PIDs of the MC very much to the point of oscilation. ANd doing the hand test i feel that the yaw is much much weaker than the roll and pitch compensation. If I crank more the YAW PIDs in order to have stronger YAW response during the transition, the quad starts to oscillate. 

This pilot mistake with under throttle in the transition I think will be a very very often made. This is because psychologically the pilot makes approach toward the transition point the same way he approaches the airstrip for landing a plane. So in most cases the pilot lowers the throttle to about 10-13% as for landing and at the spot the pilot flips the transition. As a Pilot I expect an automatic transition despite the fact I'm in Manual but in reality I don't get it. When I see the problem as a pilot in the transition in most cases I cannot recover later even with full throttle.

Not the best thing that the back transition is semi auto in Manual. But we now started making the transition in Position and Altitude. And we see the same problem but this time with just regular throttle of 50%. We just want to make the transition in full auto keeping the height. But until the machine stops fully its forward motion after the back transition we see these yaw problems and the machine is doing awkward things in MC mode after the transition. It is obvious from the second image above i sent. The second image is 10 seconds after the first image ie after the transition. We did not touch any thing just holding the RC in our hand... and it started yawing by itself with almost no horizontal speed. The weather was very very calm.

I remember how we had some problems with the front transition regarding throttle. So after some discussions with @sanderux and @LorenzMeier we perfected the front transition and since then we have 0 crash rate in the front transition. So much so that we transition with just 1.5 meter of clearance.

I think that there should be some additions to the back transition to be rock solid as well.

1. Apply preset throttle during back transition, that will guarantee enough power for good transition. After there is no horizontal speed give back the control to the pilot. (We have this in the front transition)
2. After the MC kicks in during transition, give some braking pitch down command to the MC (or pitch up the nose of the plane with the quad) in order to stop the plane faster and hover it. It should be very gentle , nothing more than 5% the most 10% pitch. After the plane stops  and hovers give me back the control as a pilot. If you decide to still give me the control during transition make it less weighted than usual control. 
3. The yaw authority should be somehow bigger during this transition. Especially when we have no Yaw control over the FW.

I will tell you a test we do and we can crash the PX4 every time in the back transition. Try making the back transition during FW roll. It doesn't matter at what throttle you do it, it is almost a certain crash.
Of course it is not meant to be this way, but people do stupid things so we have to protect ourselves making a product. The same is not true for our benchmark KK2.0 version of the VTOL. When we make the transition with the KK20 then the quad just levels the plane and handles it so much better, that very very seldom we get the leaf effect falling. The yaw handling with the KK is somehow much better and solid.

tumbili, 2016-06-08:
@tubeme Thanks for the detailed explanation. Looking at another log of yours I see that the back-transition (2 seconds) happens without any issues but then your are commanding roll angles while the vehicle is still travelling with a high speed (10 m/s). In that situation only the multicopter position controller is active and you are trying to manoeuvre it around. Therefore, I'm not surprised that unexpected things are going to happen. We may need support braking during the transition and I think that @AndreasAntener was intending to work on this.
Also I think the user rc input during a transition is a good point which you mentioned. We are going to investigate that and decide how we are going to handle user inputs in the future also considering different modes.
From our experience we can just say the following: Just flick the transition switch and don't touch the sticks until the plane reduced it's velocity. Of course I understand that a user may be tempted to do something else.

sanderux, 2016-06-09:
@tubeme 

>> Have you tried out the VTOL weathervane strategy?
> This is not applicable to our VTOL, because it is flying wing design. We get YAW control only after we make the transition from the quad.

Assuming you flying wing has stabilizers (most likely at the wing tips) this will indeed help your performance.

tubeme, 2016-06-10:
@tumbili Hi Roman

> Just flick the transition switch and don't touch the sticks until the plane reduced it's velocity.

This is exactly what we did in the first log I posted. It was in Position mode. and here is how it looks.

[image]

Also our other system is with KK2.0 In it we have control only by the quad during transition ie similar scenario. And we can safely perform transition in roll with it front and back. 

May be it is our YAW PIDs. Can you spot something not right?

MC_YAWRATE_D	0.018099999
MC_YAWRATE_FF	0.0
MC_YAWRATE_I	0.7
MC_YAWRATE_MAX	200.0
MC_YAWRATE_P	0.4
MC_YAWRAUTO_MAX	45.0
MC_YAW_FF	0.7
MC_YAW_P	3.2

The frame is flying well, it handles heavy transitions with its little brain of KK2.0. It should be a kids game for the power of Pixhawk to handle it properly... Is it the EKF2? Are we with EKF2 by default on VTOL MC mode?

In Manual mode we have e very good transition front and back...

tubeme, 2016-06-10:
@sanderux I realize that we have winglets that will help in weathervane mode we will try it in later stage of our research. For now we want to see the real handling of the quad in stronger weather.