PX4/PX4-Autopilot #4765: SITL takeoff problem

marianobianchi, 2016-06-07:
When running "make posix_sitl_default gazebo" and then execute "commander takeoff 1" in the px4 console everything works correctly.

On the other hand, if I run these commands in different consoles:
* Console 1:
  * source <ROS_WORKSPACE>/devel/setup.bash
  * source <FIRMWARE_REPO_PATH>/integrationtests/setup_gazebo_ros.bash <FIRMWARE_REPO_PATH>
  * roslaunch px4 posix_sitl.launch
* Console 2:
  * source <ROS_WORKSPACE>/devel/setup.bash
  * roslaunch mavros px4.launch fcu_url:=udp://localhost:14540@localhost:14557
* Console 3:
  * source <ROS_WORKSPACE>/devel/setup.bash
  * rosservice call /mavros/cmd/arming true
  * rosservice call /mavros/cmd/takeoff 0 0 0 0 1

I expected to have the same result as running "takeoff" in the px4 console but instead the UAV took off inclined to one side. Then it crash in the ground and stayed upside down. I attach some photos to clarify my poor english (sorry!)

[image]

[image]

[image]

Am I doing something wrong?

LorenzMeier, 2016-06-08:
I think the culprit is your takeoff command. You're telling the drone to fly to africa (latitude and longitude 0, 0). You need to send it either NaN to ignore those or send the current position.

closed 2016-06-08 by LorenzMeier

marianobianchi, 2016-06-08:
Ohh, i'm sorry... I thought /mavros/cmd/takeoff was a takeoff at current position (using current position as 0,0 for longitude, latitude).

I will try to learn how to debug so i don't bother you with these kind of questions.... If you have some advice on how to debug I will thank you pretty much

marianobianchi, 2016-06-10:
@LorenzMeier, I tried running "rosrun mavros mavcmd takeoffcur 0 0 1" to avoid giving current coordinates. I get a message "Taking-off from current coord: Lat: 47.397742 Long: 8.5455941" but I still get the same behaviour as before.