PX4/PX4-Autopilot #4766: No throttle response in manual without position publication

jgoppert, 2016-06-07:
We need to investigate further, but while testing LPE, if it wasn't publishing position due to waiting on a GPS signal for altitude initializatoin (LPE_GPS_ON) flag, the throttle of the vehicle would not respond even when in full manual control.

LorenzMeier, 2016-06-08:
@julianoes Can you have a look?

mhkabir, 2016-06-08:
This is because of land detector. It doesn't (recently) work without the position estimate.

julianoes, 2016-06-08:
Thanks for reporting @jgoppert. I'm looking into it.

julianoes, 2016-06-08:
I can reproduce the problem, it's only related to LPE and doesn't happen with ekf2. Will debug further.

julianoes, 2016-06-08:
@jgoppert I did a git bisect and it points me here: 97e6a57501e58b3151300dcaa8d7215d9a47eb9c

As I understand it, the following happens:
1. Local position is not published by LPE without GPS
2. The position control loop doesn't run because it does `continue` [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1239), so it doesn't pass on the manual input.
3. The attitude control loop doesn't get manual input.

@tumbili : Why does the manual input need to go through the position controller in manual mode?

tumbili, 2016-06-08:
@julianoes Some time ago the attitude setpoint was generated both in the position controller and in the attitude controller. After that we moved the setpoint generation into mpc only. The cleanest way would probably be to have the stick mapper module publish the setpoint but that is nowhere near to be completed.

tumbili, 2016-06-08:
@jgoppert @julianoes Why doesn't LPE publish local position without gps? It can still use baro for height.

jgoppert, 2016-06-08:
@tumbili LPE has a flag LPE_GPS_ON, if it is set to 1, it expects that a GPS is connected, and will wait to initialize the altitude until enough GPS readings have been received. If LPE_GPS_ON = 0, then it will allow baro to initialize the altitude. The issue was that the baro initialization was typically happening first and all sensor offsets were being calculated relative to baro instead of GPS which is generally more accurate.

The main issue is here, you should still be able to fly in manual without LPE or EKF2, or any position estimator at all publishing data.

julianoes, 2016-06-09:
@jgoppert: so where does this need to be resolved? Do you think that manual flight should be allowed even if no local position is published?
This would mean that the input topic copying needs to be happening differently. The easiest would be to remove the `continue` in the position controller and make the poll timeout smaller.

jgoppert, 2016-06-09:
@julianoes I think that manual control should be possible without a position estimator running, so whatever is easiest to make that work.

LorenzMeier, 2016-06-09:
Yea, short timeout plus no continue makes sense. Eventually this should go into the stick mapper.

julianoes, 2016-06-10:
Done in #4793, please review and cross-test. I hope it doesn't cause other regressions that we didn't think of.

julianoes, 2016-07-13:
Fixed in #4793, closing.

closed 2016-07-13 by julianoes