PX4/PX4-Autopilot #4767: Out of Tree Build Fails to Generate Initial/Default Parameters

sgstreet, 2016-06-07:
Building the px4/Firmware project out of the source tree generates a valid executable (successful boot) but a bad/incomplete romfs.  Using the following (assuming PX4_SRC_PATH and INSTALL_ROOT are different)

    cmake $PX4_SRC_PATH -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX=$INSTALL_ROOT -DCONFIG=nuttx_px4fmu-v4_default
    make

When the resulting image is loaded on a pixracer the following console output (partial inline, full attached) is produced:

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
param: selected parameter default file /fs/mtd_params
rgbled: no RGB led on bus #1
blinkm: I2C init failed
blinkm: init failed
param: Error: Parameter SYS_AUTOCONFIG not found.
param: Error: Parameter SYS_USE_IO not found.
param: Error: Parameter INAV_LIDAR_ERR not found.
param: Error: Parameter SYS_AUTOSTART not found.
param: Error: Parameter SYS_AUTOSTART not found.
param: Error: Parameter SYS_AUTOSTART not found.
param: Error: Parameter SYS_AUTOSTART not found.
param: Error: Parameter SYS_AUTOSTART not found.

From these error messages there appears to be a problem generating the initial/default parameter setup for the image during the out-of-tree build.

Further the current build of QGroundControl 2.9.7b does allow the vehicle to be configured in this state and board reflash is required to recover.

[Full Console Log](https://github.com/PX4/Firmware/files/303724/boot-console.txt)

sgstreet, 2016-06-08:
See pull request https://github.com/PX4/Firmware/pull/4781

dagar, 2016-06-09:
Fixed as of https://github.com/PX4/Firmware/commit/039d5528ddfba211d07d3e970d0e6d2922ea1765

closed 2016-06-09 by dagar

kd0aij, 2016-06-09:
doesn't work for me using Eclipse/Mars. Lots of errors like:
```
error finding param: VT_WV_LND_EN
error finding param: VT_WV_LTR_EN
error finding param: MIS_LTRMIN_ALT
error finding param: MIS_YAW_TMT
error finding param: MIS_YAW_ERR
error finding param: VT_WV_LND_EN
error finding param: VT_WV_LTR_EN
error finding param: MIS_TAKEOFF_ALT
```