PX4/PX4-Autopilot #4780: Primary compass not recognized upon boot

Aperture-Laboratories, 2016-06-08:
Firmware: Latest master as of yesterday (6/7/2016)
FMU: v2
Compass Hardware: Zubax GNSSv2, 3DR GPS+Compass, Internal Compass

While a compass is connected to the Pixhawk FMUv2 (whether it be a Zubax GNSSv2, 3DR GPS + Compass, or internal compass) the error reported by QGC is that "Sensors are not setup correctly" and the nutshell command "commander check" reveals this:

```
 WARN [commander] Warning:Primary compass not found
 WARN [commander] Prefilght check: FAILED
 WARN [commander] Warning: Primary compass not found
 WARN [commander] Prearmvcheck: FAILED
```

Output from top:
``` 
PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                    7004 70.715     0/    0   0 (  0)  READY 
   1 hpwork                        128  1.789   596/ 1592 192 (192)  w:sig 
   2 lpwork                         44  0.282   572/ 1592  50 ( 50)  READY
   3 init                         1280  0.000  1452/ 2496 100 (100)  w:sem 
  90 gps                            13  0.000   940/ 1192 220 (220)  w:sem 
 118 sensors                       348  2.730  1616/ 1992 250 (250)  w:sem 
 120 commander                     560  4.896  3212/ 3592 140 (140)  w:sig 
 121 commander_low_prio            505  0.000   796/ 2992  50 ( 50)  w:sem 
 126 px4io                         284  3.201  1028/ 1392 240 (240)  w:sem 
 136 mavlink_if0                    97  0.753  2308/ 2792 100 (100)  READY 
 137 mavlink_rcv_if0                 2  0.000  1308/ 2096 175 (175)  w:sem 
 144 mavlink_if1                    74  0.659  2308/ 2792 100 (100)  READY 
 145 mavlink_rcv_if1                 1  0.000   980/ 2096 175 (175)  w:sem 
 157 uavcan                        151  1.506  1132/ 2392 240 (240)  w:sem 
 174 nshterm                         1  0.000   664/ 1496  70 ( 70)  w:sem 
 187 top                           168  3.295  1252/ 1696 100 (100)  RUN   
 220 attitude_estimator_q          303  5.838  2052/ 2496 250 (250)  w:sem 
 228 position_estimator_inav        56  0.847  4484/ 5296 250 (250)  w:sem 
 234 mc_att_control                143  2.542  1180/ 1496 250 (250)  w:sem 
 237 mc_pos_control                  3  0.000  1156/ 1896 250 (250)  w:sem 

 243 navigator                       1  0.000   804/ 1296 105 (105)  w:sem
```

Output from free:
```
             total       used       free    largest
Mem:        235552     179008      56544      49968
```

@pavel-kirienko

sanderux, 2016-06-08:
I had this issue +- 2 weeks ago on master, i assumed it was bad wiring. Different branch did however seem to load without problems. @julianoes @LorenzMeier did something change here?

mhkabir, 2016-06-09:
Does this issue persist even after recalibration? 

Can you confirm that the calibration step  actually completes properly? You should reset everything and try recalibrating. Dump the pre and post calibration params here.

Aperture-Laboratories, 2016-06-09:
Saved them as text files as github doesn't support the params extension. I re-calibrated and the issue would still arise after a reboot. Every reboot I'd need to recal.

[initial_params.txt](https://github.com/PX4/Firmware/files/307092/initial_params.txt)
[post_campass_calibration.txt](https://github.com/PX4/Firmware/files/307093/post_campass_calibration.txt)

Important detail: Calibration succeeds and I can see in the Analyze window, with heading selected, that  the Zubax/3DR compass is responding (rotating it independent of the airframe).

Aperture-Laboratories, 2016-06-13:
@mhkabir Are these the params that you wanted?

telepoiss, 2016-06-26:
I seem to have the same issue with the Zubax GNSS2. 
Calibrated compass several times, tried different mounting points but still keep getting "sensors not set up correctly error"

[13:19:14.018] Info: [inav] GPS signal found
[13:19:15.124] Info: [inav] init ref: 59.4427619, 24.7337152, 16.7010
[13:19:16.648] Critical: Not ready to fly: Sensors not set up correctly

mhkabir, 2016-07-05:
OK, I can reproduce this now. Tracking.

LorenzMeier, 2016-12-28:
Should be fixed on current master. Works for me.

closed 2016-12-28 by LorenzMeier