PX4/PX4-Autopilot #4784: VTOL_LAND ignores altitude

sanderux, 2016-06-09:


tubeme, 2016-08-24:
This is very important aspect of the back transition. Often we fly high in our missions about 400 - 500 meters. In order to make a good landing in VTOL we have to maka a cascade of points or approach lowering the altitude to 20 meters and then the last point should be VTOL landing. Then it is OK. But very annoying making this approach. 

It is much better if we have a last point let say 500 meters high and then make a VTOL landing spot. The strategy could be reach last point and enter landing mode, make a spiral to 20 height then straighten flight to landing spot and make transition.

tubeme, 2016-08-24:
Another aspect of landing in our ~70 missions on 1.3.4 was that it always overshoots with 10-15 meters from the spot. Then after back transition it starts drifting back on quad. You know that at certain conditions going backwards with airplane is not its nature. 

We did not tried yet the weathervaning. If it is ON what is the strategy to go to the landing spot after back transition? Does it face the landing spot, or it faces the wind and go in the direction of landing spot?

AndreasAntener, 2016-08-25:
@tubeme you can achieve the same thing with just one additional waypoint that is on 20m just before the VTOL Land waypoint (for now).

tubeme, 2016-08-26:
@AndreasAntener That is what we are doing. Just making the last point before landing at 20m.

closed 2017-04-12 by dagar

Hylke-Atmos, 2021-02-08:
I'd like to reopen/revisit this issue!

I've noticed that both the VTOL_LAND as well as the VTOL_TAKOFF commands do not comply with the MavLink protocol.
They do not comply because;
- In the first place simply because if I'm sending a mission item with certain parameters to the vehicle I expect them to be returned when I request the mission list. Which is not the case, for example the "Approach altitude" (param3) in VTOL_LAND and "Transition Heading" (param2) in VTOL_TAKEOFF command are ignored during parsing.
- Due to the fact that it s not parsed, I'm also sure they are not used and as stated by both @AndreasAntener and @sanderux there are some very good arguments why "Approach altitude" should be handled correctly.

Yes, there are always **workarounds** like the suggested additional waypoint on top of the landing spot, but those should not be considered solutions, since have their own drawbacks (such as undesired overshoot, etc).

In regard to the "Transition Heading" I believe [issue 12455](https://github.com/PX4/PX4-Autopilot/issues/12455) is related, but only seems to be solved in PX4 Code and not in Mavlink interface.