PX4/PX4-Autopilot #4788: pixhawk v.s. pixracer

samuel6113, 2016-06-09:
Hi all, I have a question about the difference between pixhawk and pixracer. I want to install lidar lite and px4flow on my platform and use mavros to echo their topics. I was able to do this on pixhawk but not on pixracer. I assume most of the hardware setup between them are the same except pwm pins. The firmware version is also different (v2 v.s. v4). Is there anything I need change in the firmware to establish the communication? 

P.S. I was able to read data from NuttX shell using 'px4flow start' and 'px4flow test' on pixracer via I2C. Thanks!

dagar, 2016-06-09:
It's only started on the px4fmu-v2 and MINDPX_V2 at the moment.
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L594

@LorenzMeier is there a reason px4flow shouldn't be started for all configs? Perhaps gated by a param like SENS_EN_SF0X.

LorenzMeier, 2017-04-30:
This is in.

closed 2017-04-30 by LorenzMeier