PX4/PX4-Autopilot #4791: USB connection hangs on OS X El Capitan

DonLakeFlyer, 2016-06-09:
The app will hang during serial port open. This happens with QGC but has nothing to do with QGC. It can be reproduced with a simple C application that just opens the serial port (or connecting with a terminal app). If you connect to the boot loader the serial port open will succeed. If you open the port after the main firmware is up and running it will hand for a minute or so and finally succeed.

Related QGC Issue: https://github.com/mavlink/qgroundcontrol/issues/3225

dogmaphobic, 2016-06-10:
Note that it might not happen the first time around. If you disconnect (close) QGC and start again (without restarting the Pixhawk), it will eventually happen. Once it does, it will stay in that mode until you restart the Pixhawk.

DonLakeFlyer, 2016-06-10:
I can try again but I'm pretty sure I can repro just using a terminal app and no QGC at all ever.

DonLakeFlyer, 2016-06-21:
Nope I can't repro without getting QGC involved. You need to connect using QGC once. Then the port is out of whack even without QGC running.

LorenzMeier, 2016-06-21:
@DonLakeFlyer Its possible its somewhere an issue deep in NuttX, but what it really means is that the serial port handling in QGC is triggering it. Is there a way for us to simplify our serial open handling?

DonLakeFlyer, 2016-06-22:
Doing some ad-hoc testing:
- PX4 Flow connection does not hang
- UBlox RTK GPS connection does not hang

anton-matosov, 2016-11-16:
I get this problem for SITL+gazebo. Simulator hangs in src/modules/simulator/simulator_mavlink.cpp on the call to `openUart`. I have tried Pixhawk and Pixracer, hangs in both cases and hangs forever.

dagar, 2017-04-11:
Any better?

DonLakeFlyer, 2017-04-11:
Fixed ages ago