PX4/PX4-Autopilot #4794: MAVLink stream parameter

sirPerna, 2016-06-10:
It would be great to have a parameter to set Mavlink data rate. Also is the system time is now active by default. Is it possible to add this. This way you could set the MAVLink rate according to your telemetry radio. 

Right now I handle it by extras file:

mavlink stop-all
mavlink start -d /dev/ttyS2 -b 57600 -r 4000
mavlink stream -d /dev/ttyS2 -s ATTITUDE -r 15
mavlink stream -d /dev/ttyS2 -s GPS_RAW_INT -r 5
mavlink stream -d /dev/ttyS2 -s SYS_STATUS -r 5
mavlink stream -d /dev/ttyS2 -s SYSTEM_TIME -r 1
mavlink start -d /dev/ttyS1 -b 57600 -r 4000
mavlink stream -d /dev/ttyS1 -s ATTITUDE -r 15
mavlink stream -d /dev/ttyS1 -s GPS_RAW_INT -r 5
mavlink stream -d /dev/ttyS1 -s SYS_STATUS -r 5
mavlink stream -d /dev/ttyS1 -s SYSTEM_TIME -r 1

LorenzMeier, 2016-06-21:
@sirPerna I suggest you remove this and send it from the radio module now instead. See this change which went in:
https://github.com/PX4/Firmware/pull/4832

You can now send a COMMAND_LONG message with cmd = MAV_CMD_SET_MESSAGE_INTERVAL, param 1 = message ID and param 2 = interval in microseconds (not milliseconds!)

sirPerna, 2016-06-21:
Can't get it working yet. I send a frame with the information below to activate the system time frame.

static mavlink_command_long_t mvl_command_long; 

mvl_command_long.command = MAV_CMD_SET_MESSAGE_INTERVAL; 
mvl_command_long.confirmation = 0; 
mvl_command_long.param1 = MAVLINK_MSG_ID_SYSTEM_TIME; 
mvl_command_long.param2 = 1000000.0; 

target_system and target_component are the IDs I'm receiving. 

Is ist also possible to change the global data rate like when I start mavlink?
mavlink start -d /dev/ttyS2 -b 57600 -r 4000

MHageH, 2016-06-21:
Are you sure you want to set confirmation to 0 ?
This should work tho :
```
int change_rate( uint32_t rate ){
	mavlink_command_long_t change_rate_command = { 0 };
	change_rate_command.target_system    = system_id;
	change_rate_command.target_component = autopilot_id;
	change_rate_command.command          = MAV_CMD_SET_MESSAGE_INTERVAL;
	change_rate_command.confirmation     =  true;

	change_rate_command.param1            =  MAVLINK_MSG_ID_SYSTEM_TIME; 
    change_rate_command.param2            =  rate; // microseconds

	// Encode
	mavlink_message_t message;
	mavlink_msg_change_rate_command_long_encode(system_id, companion_id, &message, &change_rate_command);

	// Send the message
	int len = serial_write_message(message);

	return len;
	}
```

sirPerna, 2016-06-21:
I followed this:
https://pixhawk.ethz.ch/mavlink/#COMMAND_LONG
confirmation	uint8_t	
0: First transmission of this command. 1-255: Confirmation transmissions (e.g. for kill command)

sirPerna, 2016-06-21:
Got it working! Also data rate can now be changed by parameter 3 
2f113a4ce3138d3ec9b595636aa0656b47a31ebf

closed 2016-06-21 by sirPerna