PX4/PX4-Autopilot #4795: TBS Caipi parameters

sirPerna, 2016-06-10:
I tested PX4 on my caipi now for a few months and it works great! The attached parameters worked best for me. Is it possible they become the default parameters for Caipi flying wing? I use the 3s power setup. 
[Caipi PX4.txt](https://github.com/PX4/Firmware/files/309159/Caipi.PX4.txt)

dagar, 2016-06-10:
Can you put the important params you want to change in this file https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/3100_tbs_caipirinha and post a test flight to logmuncher?

http://logs.uaventure.com/

sirPerna, 2016-06-18:
How can I reverse PWM outputs? I added this:
 	param set PWM_MAIN_REV0 1
  	param set PWM_MAIN_REV1 1
It doesn't seem to accept it.

LorenzMeier, 2016-06-18:
Set them via QGroundControl instead. Make sure to not have any text files on your SD card.

sirPerna, 2016-06-18:
So it's not possible to reverse the PWM outputs modifying following file? 
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/3100_tbs_caipirinha

LorenzMeier, 2016-06-18:
ah, sorry, in the autostart file its possible and correct.

closed 2016-06-23 by LorenzMeier

IeiuniumLux, 2018-07-05:
In v1.8.0 the PWM_MAIN_REV* params are not longer listed/displayed in QGC v3.3.2.  Does this mean that we now need to manually set them up in [the init file](https://github.com/PX4/Firmware/blob/7a82c777b28c49b3476be3da716984e6bdc987a9/ROMFS/px4fmu_common/init.d/3033_wingwing) and build the firmware?

LorenzMeier, 2018-07-05:
They should load in QGC. Did you look in the PWM Outputs group?

IeiuniumLux, 2018-07-05:
Yes, I did.  Only the AUX and DIS are listed.

PR #4883 (2016-06-23): Update default Caipirinha parameters
fixes  #4795