PX4/PX4-Autopilot #4796: COM_FLTMODE1 uses "Hold" instead of "Loiter" in meta data

DonLakeFlyer, 2016-06-11:
I assume that "Hold" is what used to be called Loiter. QGC has been calling this Loiter forever. As does various other parts of the PX4 docs. Should this be Loiter? Right now it is inconsistent between QGC and firmware.

DonLakeFlyer, 2016-06-11:
Same for all other FLTMODE# params.

DonLakeFlyer, 2016-06-11:
Many issue here: parameter meta data, confusing doc with another different set of names. Start reading the PixRacer RCGroups thread from here:

dagar, 2016-06-11:
@DonLakeFlyer your comment above is missing the rcgroups link

DonLakeFlyer, 2016-06-11:
Updated

dagar, 2016-06-11:
Wow there's a lot of confusion out there over PX4 modes, especially multimode RC selection.

@DonLakeFlyer @LorenzMeier is the plan to completely transition to the simple flight mode selection someday?

LorenzMeier, 2016-06-11:
Just default users to simple.

DonLakeFlyer, 2016-06-11:
Already does that for new users wih default param set. Even in single channel you run into problems with things like Manual and Stabillize being listed individually in the dropdowns. As well as the naming differences.

dagar, 2016-06-12:
For mavlink custom mode why bother with the submode? What if we just started using nav_state as the custom mode?
https://github.com/PX4/Firmware/blob/master/src/modules/commander/px4_custom_mode.h#L57

DonLakeFlyer, 2016-06-13:
The other confusion is Manual and Stabilize both being listed separately.

dagar, 2016-06-13:
Yes, that's a frequent point of confusion.
Manual and stabilized are unique states, but for multicopter the controllers are doing the same attitude and rate stabilization. We could either block manual for MC, or requesting manual puts you in the actual stabilized main state.

On the QGC side a few states are missing like rc recover, engine failure, etc.

Can we tolerate changing px4_custom_mode at this point? We could get rid of the confusing union and just make the px4_custom_modes line up with nav_state. For requesting modes we could even make nav_state a superset of commander_state to keep it simple.

DonLakeFlyer, 2016-06-13:
> Can we tolerate changing px4_custom_mode at this point?

That's going to blow back combat out of the water.

dagar, 2016-06-14:
I made the naming consistent. The only outstanding piece is the documentation growing rather out of date...
http://dev.px4.io/concept-flight-modes.html

DonLakeFlyer, 2016-06-15:
So instead of Loiter the name is Hold. Yet the name for the Mission Commands (per mavlink spec) are Loiter. Not sure that makes sense?

LorenzMeier, 2016-06-15:
The HTML code for a link is `href`, does that mean internet users have to call a link "href"?

dagar, 2016-06-15:
Flying a hold (holding pattern) is standard in aviation, so I think it makes a good mode name. If not sure if it makes it more or less confusing to name the mode the same as the waypoint type.

DonLakeFlyer, 2016-06-15:
I don't really care what the name is. I care about it being called essentially two different things in two end-user visible places. If nobody else cares, then neither do I.

closed 2016-06-16 by dagar