PX4/PX4-Autopilot #4797: Check roll direction for VTOL

LorenzMeier, 2016-06-11:
I have the odd symptom that the ailerons signs are inverted for VTOL in SITL.

LorenzMeier, 2016-07-29:
Will fix this in v1.5

dagar, 2017-02-18:
How do we transition in a fix like this?https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/standard.cpp#L439

LorenzMeier, 2017-02-19:
I think we have no chance but to bundle this into a major release with big red warnings.

sanderux, 2017-04-02:
Following up here on the inversion issue of the roll on vtol's:
https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/standard.cpp#L456

@dagar commented:
> This could be a good excuse to unify the VTOL mixers with regular FW & MC. Some of them are special, but many could just reuse a regular MC mixer on MAIN, regular FW mixer on AUX.

As far as the DeltaQuads go i would actually prefer using only MAIN outputs
This will make the setup compatible with FC's that do not provide an AUX rail

dagar, 2017-04-02:
Nevermind, I just realized I don't know what I'm talking about. The current VTOL FW mixers are control group 1.

mhkabir, 2017-04-23:
@bresch Can you please have a look at this.

LorenzMeier, 2017-04-23:
Changing this for the next release as a breaking change - not for this one.

LorenzMeier, 2017-10-05:
Needs an API compatibility system that checks versions in params, compares it with the version in code and locks down arming if they mismatch or are smaller. Also should have a proper UI representation that includes instructions and a link to a more in-depth explanation on the user guide.

bresch, 2018-02-21:
Still have to fix that. Maybe with the refactoring of VTOL @RomanBapst

dagar, 2018-02-21:
@RomanBapst we should definitely fix this.

RomanBapst, 2018-02-26:
@bkueng and I have decided to finally tackle this one.
Here is our suggestion so far:

- we add a line of code into all mixer files supported by PX4 which can be parsed by the mixer, e.g.
V: 2
- if the mixer does not find this line it returns an error which can be handled accordingly by the system
- handling of the error message could include refusing to arm and sending a message to QGC (this is where input from others would be appreciated!)

This approach makes sure that no user can fly with a custom mixer without adding the magic line and thereby understanding that he/she requires to update their custom mixer files.

Please provide feedback!

dagar, 2018-02-27:
Yes I think treating it like adding simple schema versioning would help us solve this annoyance as well as handle other changes in the future.

The other idea I had was to introduce a series of interactive preflight check walkthroughs, but that would need a lot more thought (and effort). On a newly configured airframe or major firmware upgrade we could force the checks to be rerun (unless the user explicitly opted out).

LorenzMeier, 2019-02-03:
@RomanBapst I've assigned this to you for triage. My preference would be to fix it now before even more people fly VTOL and it will become impossible to change.

jlecoeur, 2019-02-03:
@dagar OK I can give it a go. I just need your help to know where to best implement the arm/preflight check.

PR #11382 (2019-03-15): VTOL: fix roll inversion
fixes #4797 

**Describe problem solved by the proposed pull request**
The VTOL attitude controller generates reversed fixed-wing roll commands. As a consequence, all VTOL mixers currently have a minus sign on roll actuators to compensate.

**Describe your preferred solution**
- Fix sign of the fixed wing commands generated by the VTOL attitude controller
- Fix builtin mixers
- Add a parameter to force users of custom VTOL mixers to check roll control surfaces before flight
- Fix builtin airframes scripts to bypass the new parameter

**Flight testing**
This requires ex [... cut]
closed 2019-03-15 by dagar