PX4/PX4-Autopilot #4799: BAT_V_SCALING and BAT_C_SCALING disappeared w/o ROMS vehicle updates

jgoppert, 2016-06-11:
I just tried to fly my 3dr iris on master, but the battery voltage is reading 10.2 when it should be 12.5 volts. I looked at the firmware on master and saw that all of the params set fot he battery in the iris ROMFS have now disappeared.

dagar, 2016-06-12:
BAT_V_SCALING -> BAT_CNT_V_VOLT
    BAT_C_SCALING -> BAT_CNT_V_CURR
    BAT_C_OFFSET -> BAT_V_OFFS_CURR

I've seen this happen a few times so I was trying to think of something we could do to catch it. Maybe we could default `param set` to fail if not found. Then if we had a nuttx sim build it could run through loading all airframe configs. The other option is just being more careful...

jgoppert, 2016-06-12:
Yeah, sending a message to the use would be best via mavlink info.

jgoppert, 2016-06-13:
Resolved for me on master.

closed 2016-06-13 by jgoppert