PX4/PX4-Autopilot #4800: SITL targets confusion

julianoes, 2016-06-12:
Some of the SITL targets don't seem to be accessible to run anymore, e.g.

`make posix_sitl_lpe gazebo`

We should go through all of them and consolidate, fix them or clean it up.

mhkabir, 2016-06-12:
While not strictly related, this might be the time to start consolidating startup scripts on POSIX. Like we have for NuttX, having a core "rcS" script, and calling auxiliary scripts depending on vehicle type, or SITL mode etc. would be nice. 

I'm not sure if the PX4 shell can handle this yet. I would be willing to do this cleanup, it if someone would give me the pointers on what needs to be implemented.

julianoes, 2016-06-12:
@mhkabir I agree, but I don't know how to tackle this (yet) because the PX4 "shell" doesn't know any command except starting commands and Ctrl+C as far as I know.

I would think we need some kind of a shell compatible to the NuttShell (Or re-use the NuttShell itself?).

mhkabir, 2016-06-12:
I think @bkueng worked on the shell a bit. @bkueng could you let us know please? Thanks!

dagar, 2016-06-12:
I believe he added colors, comment handling (#), and some other fixes. To unify the init scripts we'd need env variables, conditionals, the ability to call another script, and a few other little things. You could start faking it by implementing an `sh` that executes another list of commands. Then you'd be able to get rid of the bulk of the duplication.

bkueng, 2016-06-13:
I'm all for consolidating as well. Current Posix shell support is very limited, implementation is here: https://github.com/PX4/Firmware/blob/master/src/platforms/posix/main.cpp#L145
As @julianoes  wrote, it only supports a list of commands and handles comments (colors are indepentent from the shell).

A quick look at NuttX shell shows that it's quite interleaved with NuttX, so I don't think we can easily pull it out and use it separately. But its syntax & semantics is well-documented under https://github.com/PX4/NuttX/blob/master/apps/nshlib/README.txt

julianoes, 2016-06-13:
@bkueng Thanks for checking that.

I wonder if we can somehow make use of existing shells (sh or bash) Linux systems. Can we spawn threads of mainapp from sh/bash (or trigger it)?

bkueng, 2016-06-13:
We could add a shell interpreter (like bash) as a library, but I did not find much about the topic. Maybe I'm lacking google skills?
You can use busybox as a library (libbusybox), which includes a shell, but it's also not really documented.

mhkabir, 2016-06-13:
Thanks for the replies everyone!

I think short-term, I will just implement a basic 'sh' functionality to the shell allowing other sub-scripts to be called from within pxh (as suggested by @dagar). That should go a long way in reducing the duplication. I will get started on this.

As the next step, we can just roll our own support for the other tidbits like conditionals, and maybe also look into this

> somehow make use of existing shells (sh or bash) Linux systems. Can we spawn threads of mainapp from sh/bash (or trigger it)?

A question I had here - How exactly does PX4 run on POSIX?  It definitely isn't task-based like on NuttX, so is it one monolithic app (mainapp) which spawns threads for each application/module?

mhkabir, 2016-06-13:
@bkueng Some googling revealed this : http://wanderinghorse.net/computing/shellish/eshell.html. Might be interesting.

I'm still searching for more options.

bkueng, 2016-06-13:
@mhkabir Yep looks interesting. But latest release dates years back. Doesn't seem to be maintained anymore.

mainapp uses threads for everything.

mhkabir, 2016-06-13:
I have a feeling that we'll eventually just roll our own shell :)

mhkabir, 2016-06-15:
Made a start here : https://github.com/mhkabir/Firmware/tree/posix_shell
This will expand to something that can handle most of NSH functions over the next weeks :) 

Is is OK to use Linux environment variables to save our internal ones? e.g MAV_TYPE, etc. ?

closed 2018-02-21 by PX4BuildBot