PX4/PX4-Autopilot #4803: Set yaw setpoints in waypoint navigation

ndepal, 2016-06-13:
PX4 supports multiple yaw modes for multicopter waypoint navigation as described here: https://pixhawk.org/users/multirotor_yaw_modes

I need finer control on the drones yaw behavior. Specifically, I need to be able to control where the drone is facing at all times. E.g. I want to be able to fly to an object and then around it, always facing it.
Thus it would be useful to be able to set a yaw setpoint in a waypoint, which is interpolated between waypoints.

As far as I can see, this is not currently possible. Is this something that could be implemented?

I suppose I could forgo using waypoints altogether and send position/yaw setpoints over mavlink with offboard control, but it seems like this would be a useful feature for waypoint navigation.

dagar, 2018-01-17:
Still needed?

ndepal, 2018-01-17:
Nope, heading towards ROI solves this for me.

closed 2018-01-17 by ndepal