PX4/PX4-Autopilot #4808: region `flash' overflowed by 3940 bytes

lovettchris, 2016-06-13:
On a clean clone of master branch,  make px4fmu-v2_default fails with error saying:

100%] Linking CXX executable firmware_nuttx
e:/px4/toolchain/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: firmware_nuttx section `.data' will not fit in region `flash'
e:/px4/toolchain/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: **region `flash' overflowed by 3940 bytes**
collect2.exe: error: ld returned 1 exit status
make[3]: *** [src/firmware/nuttx/firmware_nuttx] Error 1
make[2]: *** [src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [px4fmu-v2_default] Error 2

dagar, 2016-06-13:
It looks like you're using GCC 4.7.4. The build system uses GCC 4.8/4.9 and they're both (barely) under the 1MB max for a pixhawk.
You'll either need to update to a newer version of GCC or disable modules you don't need by commenting them out here - https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake

http://dev.px4.io/starting-installing.html

lovettchris, 2016-06-13:
That was it, thanks.

closed 2016-06-13 by lovettchris