PX4/PX4-Autopilot #4816: Mavlink forwarding not working on master

mantelt, 2016-06-14:
I'm starting to mavlink instances on px4, both with the -f option, to forward message from once instance to the other. In the current master, this does not work anymore.

I have the following configuration:

OBC (mavros) <-- (UART) --> PX4 <-- (UART/3DR) --> GCS (QGC/mavros)

On earlier releases, I could use the mavlink -f option (respectively -p and -f) to forward messages on the PX4 from OBC to GCS. As I'm not regularly upgrading to newest version, I can not point to a certain commit where it breaks. But as mavlink v2.0 is being integrated at the moment, I guess it happened recently.

mhkabir, 2016-06-14:
Can you run  ```git bisect``` and quickly see which commit breaks it?

dagar, 2016-06-14:
Diff between v1.3.1 and v1.3.2
https://github.com/PX4/Firmware/compare/v1.3.2...v1.3.1

mantelt, 2016-06-14:
@dagar There are definitely more changes from v1.3.1 to v1.3.2 than shown in the above diff.
https://github.com/PX4/Firmware/compare/v1.3.1...v1.3.2 seems to be more representative of all the commits that were added.

Although I still can't pinpoint a certain commit that breaks the functionality, I'm quite sure it's related to the introduction of mavlink v2.0 into the mavlink module (i.e. commits 6eccfe3d14febbea68d52292fd8099fdf3a806dc to 56e887c6c7599fe15349540a527a17402f0d786c in https://github.com/PX4/Firmware/commits/v1.3.2/src/modules/mavlink ), as forwarding does not work anymore on 6eccfe3d14febbea68d52292fd8099fdf3a806dc.

@mhkabir: I've never used git bisect so far, I'll try to figure out how to use it and what information I gain from it.

dagar, 2016-06-14:
Whoops, that makes more sense. Commits in v1.3.2 not in v1.3.1. I didn't expect so much change in a patch release.

git bisect works best if you can automate a simple test, which might be hard in this case.

mantelt, 2016-06-16:
As I'm currently busy working on other things it will be a while till I may have another look on this issue to narrow it down further. I would of course appreciate it a lot if somebody else could jump in and investigate...

PR #4907 (2016-06-25): Fix mavlink forwarding issue.
Fix #4816 Mavlink forwarding not working on master.

PX4 master introduce mavlink v2.0 from v1.3.2, but mavros still use mavlink v1.0. 
It will cause compatibility problem when forwarding mavlink v1.0 message.
[MAVLINK_NUM_HEADER_BYTES](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1025) always represents mavlink v2.0 header bytes regardless of mavlink v1.0 or v2.0 message to forward.

[mavlink api _mavlink_resend_uart](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L355) can distinguish v2.0 from v1.0.
closed 2016-06-25 by LorenzMeier