PX4/PX4-Autopilot #4819: Low altitude take-off overshoot

jgoppert, 2016-06-14:
When doing auto missions with LPE I noticed this:

## Altitude spike at start of mission

Video:
https://www.youtube.com/watch?v=t7s8027uKZ4

Log:
http://logs.uaventure.com/view/beRXQ3uES9ctPkv2H5ymDQ

I think it is related to the takeoff command, the z setpoint is not showing it, even though the climb does appear in agl, this is in MPC_ALT_MODE = 1 so the controller is tracking AGL.

```python
pl.figure(figsize=(15,5))
(data.EST0_s7 - data.EST0_s2).plot(label='AGL, m')
(-1*data.LPSP_Z).plot()
data.STAT_MainState.plot()
pl.grid()
pl.legend(loc='best')
```

[image]

dagar, 2017-04-11:
Was this ever resolved?

MaEtUgR, 2017-04-18:
https://github.com/PX4/Firmware/pull/7061 related?

Stifael, 2017-04-19:
So I have kind of first solution, but need to check for corner cases since the entire auto takeoff is kind of a mess. 
@LorenzMeier: there is however still an issue (also with the current master), that the mc_pos_control and the navigator are not in sync. this means that the mc_pos_controller finishes its takoff, but the navigator remains in the takoff mode. that means that the user has to switch to a new mode to continue

LorenzMeier, 2017-04-19:
That's not actually the case. If you do a takeoff as an action staying in thst mode is fine and in fact for fixed wing more desirable.

If you do it as Auto mission it will continue with the next item.

We need to focus here on fixing the takeoff ramp / state machine. One thing at a time and iterate fast.

Stifael, 2017-04-19:
@MaEtUgR will take over

MaEtUgR, 2017-04-27:
This is potentially related...
I made a predictable smooth takeoff logic without special magic thrust generation in the begining, it's coming in here: https://github.com/PX4/Firmware/pull/7138

While testing the feature I realized that there are situations where it still jumps and that it's most probably not related to my changes. I isolated the problem to be in the auto takeoff logic. When you log with the new logger (ulg) you have the pos_setpoint_tripplets and see the problem:
[image]
There is a valid triplet loiter point just at the beginning before the takeoff that is usually higher then the actual takeoff point. The behaviour is then pretty unpredictable for me... and I also didn't yet find out why this would happen.

LorenzMeier, 2017-05-03:
Not critical enough to gate the release.

MaEtUgR, 2017-05-09:
I think @Stifael found the root cause for the problem I explained earlier (https://github.com/PX4/Firmware/issues/4819#issuecomment-297688753). The glitch always happens when you switch from manual to auto, the navigator and mc_pos_controller are out of sync at the beginning which could lead to this spike. Maybe you can test after we have a fix.

Stifael, 2017-05-15:
this should now be resolved. It was a general problem when switching from manual to auto. 
If the problem still exists, we can reopen it.

closed 2017-05-15 by Stifael