PX4/PX4-Autopilot #4822: ERROR: FastRpcWrapper Not Initialized

SamChenzx, 2016-06-14:
Today I followed this [guide](http://dev.px4.io/advanced-snapdragon.html#upgradingreplacing-the-linux-image): http://dev.px4.io/advanced-snapdragon.html#upgradingreplacing-the-linux-image to upgrade my  new two Snapdragon flight P2 board from Intrinsyc. With the latest master, I encountered this issue and can't calibrate sensors.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

mainapp starting.

/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:136:failed to create tls key/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:112:listener using ion heap: -1
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:225::error: -5: 0 == (nErr = dlerr)
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:228:adsp dlopen error: px4muorb undefined PLT symbol _FDtest (688) /libpx4muorb_skel.so**ERROR [muorb]    Initialize Error calling the uorb fastrpc initalize method..**
INFO  [sdlog2]   [blackbox] /root/log/sess001
INFO  [sdlog2]   [blackbox] recording: log001.px4log
**ERROR [muorb]    ERROR: FastRpcWrapper Not Initialized**
  MAV_BROADCAST: curr: 0 -> new: 1
INFO  [dataman]  Unkown restart, data manager file '/home/linaro/dataman' size is 103090 bytes
INFO  [mavlink]  mode: Normal, data rate: 1000000 B/s on udp port 14556
pxh> INFO  [mavlink]  partner IP: 192.168.1.11

pxh> WARN  [navigator] no GPS - navigator timed out

Is this a bug of snapdragon flight's DSP layer?

mhkabir, 2016-06-15:
Did you update the aDSP image?

SamChenzx, 2016-06-15:
I updated both Flight_adsp_8074_firmware_readme_update.zip
<http://support.intrinsyc.com/attachments/download/561/Flight_adsp_8074_firmware_readme_update.zip>
 and Flight_BSP_3.0_apq8074-le-1-0_r00015.zip
<http://support.intrinsyc.com/attachments/download/428/Flight_BSP_3.0_apq8074-le-1-0_r00015.zip>,
I'll
try to update it again later.

julianoes, 2016-06-15:
@SamChenzx Did you update the BSP3.0 first and then the adsp image?

julianoes, 2016-06-15:
@SamChenzx could it be that it is already (auto) started when you try to start it?

mcharleb, 2016-06-15:
@julianoes The following line would indicate you have an old adsp image that is missing _FDtest. 

adsp dlopen error: px4muorb undefined PLT symbol _FDtest (688) /libpx4muorb_skel.so

julianoes, 2016-06-15:
@mcharleb Oh yes that's right.

SamChenzx, 2016-06-15:
I updated it again with another computer today, this issue seems solved though I don't know what's wrong with my computer. I think there will be more P2 board than P1, so maybe we should adapt px4.config file to P2.

SamChenzx, 2016-06-15:
I think this board is a P2 board as it has two pins next to J3 though the silkscreen seems it's a P1 board. I tried to short those two pins to recover factory image, but failed. So it's neither a P1 nor a P2 board?
[image]

mhkabir, 2016-06-16:
It's a P1. Even the P1 has those two pins, but they're not connected to
anything.

SamChenzx, 2016-06-16:
So, actually we don't need to set any rotation parameters to the board? But
when I short circuit those two pins then power up, I can see "Detected
factory reset key" but didn't work.

mhkabir, 2016-06-16:
Everything on this board is the exact same as mine. So, I don't think you'll need the rotations.

julianoes, 2016-06-16:
@SamChenzx: so it looks like your issue is resolved. Closing this. Re-open it if it happens again. Thanks.

closed 2016-06-16 by julianoes