PX4/PX4-Autopilot #4825: Multicopter position control: Ensure that the yaw setpoint handling is against current yaw, not yaw error

LorenzMeier, 2016-06-15:
We need to ensure that the yaw transformation is done using current yaw, not the desired yaw. If that's already the case then the code needs to be properly commented to make sure anyone reading it does get it.

tumbili, 2016-06-15:
@LorenzMeier Addressed in my PR for circle mode. Not yet pushed.

closed 2016-06-15 by tumbili

devbharat, 2016-06-16:
> We need to ensure that the yaw transformation is done using current yaw

You mean transform the stick input pin position control?

What if controlling yaw while moving(or in wind) is difficult? We often see not so nice yaw control while hovering fast, but once we slow down the yaw tracking recovers. When we transform the stick input by current yaw, and hold constant pitch, it kind of curves rather than going straight. Transforming by yaw setpoint wouldn't have this problem.