PX4/PX4-Autopilot #4827: Aerocore 2 Upload Stuck at "Waiting for the Bootloader"

jmcwil, 2016-06-15:
Hi,

I am working with Aerocore 2, and I am following the Development Guide for PX4.  I have made a slight modification to the px4_simple_app so that I can verify whether my upload is working (adding more exclamation points to "Hello Sky!"), and every time I use make px4fmu-v2_default upload, I get stuck at "Loaded firmware for 9,0 size: 1020956 bytes, waiting for the bootloader... ".  I have tried unplugging and re-plugging the USB connector but have had no success.  Every time I run the version of px4_simple_app from the Nuttx Shell on the Aerocore, I find that it does not contain my modifications. Please let me know what I should do to successfully upload the firmware.

As a side note, I have tried uploading the firmware from QGroundControl, and this has been successful; however, this has not solved my problem, since I do not know how to change which firmware is being uploaded using QGroundControl, if that is possible.

Thank you,
Jessica

dagar, 2016-06-15:
We're not building aerocore binaries at the moment. It looks like aerocore never made it to the current cmake build system. I'm guessing it's not binary compatible with px4fmu-v2 (pixhawk).

@jmcwil are you interested in restoring the aerocore build? It shouldn't be too difficult, but I don't have access to this hardware to do it myself.

jmcwil, 2016-06-15:
@dagar Thanks for the response.  That is very interesting; how did you find that information?  Can you tell me how long it has been since you stopped building aerocore binaries?  I am very new to px4 and gumstix, but I would like to use PX4 autopilot on my Aerocore 2.  I am assuming that switching to px4fmu-v4 will not help, but please correct me if I am wrong.

What is involved in restoring the build? If you are able to walk me through all the steps, I would be willing to give it a try.

Alternatively, do you happen to know if this [aerocore manifest ](https://github.com/aerocore/manifest)would still be valid?  I have been unable to get it to compile on my machine so far, but it seems like it could work.

I am in the early stages of this project, so I am open to alternatives.  Any help/information is appreciated.

Thanks again!

davids5, 2016-06-15:
@jmcwil @dagar 

I think I have kept aerocore building on the nuttx_v3 branch - so it would be worth a try to see if it builds and runs from that branch master was merged into nuttx_v3 today.

dagar, 2016-06-15:
Great! If it's just a cmake/config and Makefile entry it should be easy to pull into master. It looks like the support for flashing is still in QGC, so we'd just need to get it back in the builds and uploading to s3.

jmcwil, 2016-06-16:
@davids5 
I switched to the nuttx_v3 branch and tried going through the same process to build px4fmu-v2_default, but with no success.  Is there something I might be doing wrong?  Is there some way that I can modify an existing file to make the code compatible with the aerocore 2?

Thanks for the help!

davids5, 2016-06-16:
@jmcwil @dagar No - I was wrong it is not in nuttx_v3 either. [nuttx_next] (https://github.com/PX4/Firmware/tree/nuttx_next/makefiles/nuttx) has it but that is a very stale branch and uses make not cmake. 

@jmcwil There is a pattern you can infer from looking at master, nuttx_v3 and nuttx_next and convert the old *_aerocore.mk to a cmake config for it - but it will take some work and is a great learning experience

jmcwil, 2016-06-16:
@davids5  Thanks, I think I will give it a shot, and I will definitely learn some things from this.  I am still trying to wrap my mind around the file structure, and I want to make sure I am following you.  Basically you are suggesting that I look at the nuttx_next branch .mk files in makefiles/nuttx.  Using those I should be able to make my own .cmake file for the aerocore in cmake/configs in my fork of the master.  Once I have that, I can make and upload the new file to the aerocore.  Is my understanding correct?

davids5, 2016-06-16:
@jmcwil - You are welcome. Yes more or less. But the compiler will tell you what is missing.

Just look out for this wart on master https://github.com/PX4/Firmware/blob/master/cmake/nuttx/px4_impl_nuttx.cmake#L461 
it is done better in nuttx_v3,

If I were you I would work off on nuttx_v3

Then you can submit a Pull Request and we can backport it into master.

dagar, 2016-06-16:
As a quick start you can go through the top level Makefile and cmake/ and duplicate what you see for px4fmu-v2.

you'll need a config

     cp cmake/configs/nuttx_px4fmu-v2_default.cmake cmake/configs/nuttx_aerocore-v2_default.cmake

and an entry in the top level Makefile

    aerocore-v2_default:
        $(call cmake-build,nuttx_aerocore-v2_default)

Beyond that I don't know enough about the aerocore to help you. There's `nuttx-configs/aerocore`, but is it valid for aerocore 2 as well?

jmcwil, 2016-06-24:
Hi,

@dagar @davids5 Thanks for all of the advice to get me started on a Cmake file.  There is a pretty good chance I will be doing this next week, and I think your guidance has given me a great starting point.

However, I would like to confirm that this will be my best option before I invest my time and effort into learning how to make my own cmake files.  Is there something that will prevent me from reverting to an earlier commit and flashing firmware from nuttx_next?  I am going to be making significant customizations to the software once I figure out how to upload it to my Aerocore, so I would be ok with my fork being a few commits behind.  I decided to try this approach initially, and switched branches back to nuttx_next.  However, I keep running into errors with the NuttX HEAD, and I have been unable to checkout any of the commits for the NuttX repo.  I have tried following the directions found in[ this thread](http://stackoverflow.com/questions/2155887/git-submodule-head-reference-is-not-a-tree-error), and checking out different commits found in the log, but I have been unable to get past this warning.  Is there something specific to PX4 or NuttX that could be causing this error?

In addition,  if you know a reason that would mandate my switching over to the cmake system, I would greatly appreciate an explanation.  I am new to this system, and I am still considering multiple approaches to solving my problem.

Thanks!

LorenzMeier, 2016-06-24:
The branch you describe is abandoned and will be deleted. So if you do this you essentially will never receive any updates and you will work with outdated and buggy estimators, controllers, communication, etc.

dagar, 2016-06-24:
@jmcwil I would try to get aerocore support back in master first, then we can keep it working in the build system and available through QGroundControl. From there you can cherry-pick the change into your personal fork.

I'm not sure about your specific submodule problem, but they can be a pain. Have you intentionally made any changes to a submodule like Nuttx, src/lib/ecl, etc? Your issue might just be from jumping between master and nuttx_next where the Nuttx submodule has changed to a different remote. If you haven't committed anything to the submodules themselves you can force them back in sync with `make submodulesclean`.

jmcwil, 2016-06-24:
@LorenzMeier Thanks for quick and informative response.  I see now why my other proposed method does not provide a good long term solution.  I'll get started on the .cmake file!

@dagar Thanks for the advice.  Yeah, I think my submodule problem was just caused from jumping branches.  I have gotten everything set up on the master branch again and everything is working now.

jmcwil, 2016-06-28:
Hey,
I have been working on converting old aerocore files over to cmake and working through the errors that the compiler gives me.  This one error has me pretty stuck though:
```
/home/jessica/src/Firmware/src/systemcmds/ver/ver.c:67:45: error: 'CONFIG_CDCACM_PRODUCTID' undeclared here (not in a function)
 __EXPORT const uint32_t px4_board_version = CONFIG_CDCACM_PRODUCTID;
                                             ^
```
I made sure I was up to date with the master (I know this file just got converted to cmake, and I have been getting this error before and after the update).  I looked at the line in question, and I have been unable to find out what is causing this problem.  Any ideas?

Thanks!

dagar, 2016-06-28:
That's defined in nuttx-configs. It might help to diff against px4fmu-v2 and make sure nuttx-configs is consistent.

    meld nuttx-configs/px4fmu-v2/ nuttx-configs/aerocore/

jmcwil, 2016-06-28:
Thank you for showing me where that variable is set and also for showing me how to use meld to make my life easier!  I found the differences that are causing this problem.  I am still confused about how the file /nsh/defconfig is being generated.  Its current content is incorrect- could you please show me what automatically generating it?

dagar, 2016-06-28:
I believe it's menuconfig, but I haven't used it in a long time. @davids5 probably knows more.
http://nuttx.org/doku.php?id=documentation:menuconfig

davids5, 2016-06-28:
@jmcwil what branch are you on?

jmcwil, 2016-06-29:
I decided to work on a new branch off my fork of the master.  I tried switching to nuttx_v3 as you had suggested at first, but it was causing errors with NuttX, similar to the ones I mentioned earlier in this thread.

I was able to get past the error in nuttx-configs by manually editing the file to contain the information that was missing.  Looking at the history, it appears that is what had been done in the past.

Now I am getting stopped at an error that px4_parameters.h is missing.  I have been unable to locate this file in any of the branches I have been referencing. Is there a reason it needs to be included in the systemlib/param/param.c? Or is this something I can remove?  I am confused why I do not get this error when I make px4fmu-v2_default, but only for aerocore_default.

`/home/jessica/src/Firmware/src/modules/systemlib/param/param.c:68:28: fatal error: px4_parameters.h: No such file or directory
 #include "px4_parameters.h"`

jmcwil, 2016-06-29:
UPDATE:  Nevermind about the param error, I found and fixed the problem in my CMake file.

davids5, 2016-06-29:
@jmcwil 

You can also use cmpconfig 

```cd NuttX/nuttx/tools```
```make -f Makefile.host```

```./cmpconfig    nuttx-configs/aerocore/nsh/defconfig nuttx-configs/px4fmu-v2/nsh/defconfig``` 

It is inadvisable to hand edit a defconfig file because the Kconfig tools maintains dependencies.
 (but sometimes it is unavoidable see revert to) 

For a derivative of master to get the defconfig right you can.
```make clean```
```make aerocore_default``` - it can fail that is ok
```cd build_aerocore_default/aerocore/Nuttx/nuttx``` (?)
```make oldconfig``` - take the defaults.
```make menuconfig``` - here you would visit every item on every page and compare to a  px4fmu-v2_default 

then meld .config and nuttx-configs/aerocore/nsh/defconfig and see if the changes make sense.

revert to: 
```
CONFIG_ARCH_BOARD_AEROCORE=y
# CONFIG_ARCH_BOARD_CUSTOM is not set
CONFIG_ARCH_BOARD="aerocore"
```

jmcwil, 2016-06-29:
Ok, I can try to fix the file the proper way.  Do I need to install something?  I am getting errors on make oldconfig and make menuconfig.
`/bin/sh: 1: kconfig-conf: not found
make: *** [oldconfig] Error 127`

davids5, 2016-06-29:
see https://bitbucket.org/nuttx/tools/src kconfig-frontends

jmcwil, 2016-06-29:
@davids5 I am still struggling with this part.  I followed instructions and installed kconfig-frontends to my home directory (not sure if that is where it should go).  Something with my set up must be wrong.  I am unsure of how to use this with the PX4 stuff, and I have not been able to get it to work.  I am a very new developer and am still trying to figure out what all of this means.

Can you please explain again why hand-editing the deconfig file will cause problems, or how I can tell if it has broken a dependency?  What I did is copy a section of code containing values for CDCACM from an older version of the aerocore's deconfig file that was missing from the newer version.  It seemed to work, and I did not have any more issues with it, but I realize that there could be other issues that appear later on.  What would make hand-editing unavoidable?

jmcwil, 2016-06-29:
Also, would someone mind taking a look at this separate error I keep getting?

`/home/jessica/gcc-arm-none-eabi-4_9-2014q4/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: cannot find -luavcan
/home/jessica/gcc-arm-none-eabi-4_9-2014q4/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: cannot find -luavcan_stm32_driver
collect2: error: ld returned 1 exit status
make[3]: *** [src/firmware/nuttx/firmware_nuttx] Error 1
make[2]: *** [src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [aerocore_default] Error 2`

The last section of the aerocore_default.cmake file will not properly execute.  I even tried commenting out this section to see what would happen.  I just got a similar set of errors about sercon and serdis.
`set(config_extra_libs
	uavcan
	uavcan_stm32_driver
	)`

It seems to me that I must have neglected something very important.  This last section of code is currently identical to that of the px4fmu-v2_default, yet I get no errors on its compilation.  Please point me in the right direction, if you have an idea of what is going on.

Thanks!

dagar, 2016-06-29:
You could entirely remove uavcan for now. Do you have the uavcan module in the aerocore cmake config?
https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake#L81

jmcwil, 2016-06-29:
Thanks.  I have now commented out that line as well as these:
`#set(config_extra_libs
	uavcan
	uavcan_stm32_driver
	)`
That takes care of that issue, at least for now.

Now I am left with
`````
CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj:(.rodata.g_builtins+0x2ac): undefined reference to `serdis_main'
CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj:(.rodata.g_builtins+0x2bc): undefined reference to `sercon_main'
collect2: error: ld returned 1 exit status
`````
as well as a whole bunch of errors about an ``undefined reference to `io_timers'`` in ``/drivers/stm32/libdrivers__stm32.a``.  I will keep trying to pinpoint these errors.

davids5, 2016-06-29:
> . I followed instructions and installed kconfig-frontends to my home directory (not sure if that is where it should go).

You can download and unzip it anywhere like /home/you/installs/kconfig

Then run the
```
 ./configure --enable-mconf --disable-nconf --disable-gconf --disable-qconf
    make
   sudo  make install
```
commands

> Can you please explain again why hand-editing the deconfig file will cause problems, 

Simple (somewhat artificial)  example: In the menuconfig  UART1 is selected and that enables the Console to be set, and it is set on UART1, but the operator. Later you hand edit the defconfig and set UART2 and unset UART1 - but the console remains on UART1

Hopefully you get a compiler error, if not code in ```#if defined(CONFIG_UART1) ``` will just disappear. No compiler error an boom broken system! 

>or how I can tell if it has broken a dependency?

```make oldconfig``` and ```make menuconfig```

It enforces the dependencies, and cleans out the cruft.

> What I did is copy a section of code containing values for CDCACM from an older version of the aerocore's deconfig file that was missing from the newer version. 

What about all the dependencies of the version of nuttx you are using compared to their version?

Do you have all the CDCACM options and all thier children set?

You will NOT know!

>It seemed to work, and I did not have any more issues with it, but I realize that there could be other issues that appear later on.

I am glad you recognize that. Just because it compiles does not mean it will run so "worked" is a very relative term. Compiled would be a better description. Runs and works - that will come later. :)

Just remember 30 minute is Kconfig tools is better the 3 days in a debugger. any time!
.
> What would make hand-editing unavoidable?

The Kconfig tools will remove the 
```
CONFIG_ARCH_BOARD_AEROCORE=y
# CONFIG_ARCH_BOARD_CUSTOM is not set
CONFIG_ARCH_BOARD="aerocore"
```
Because the areocore board is not defined in the Nuttx/nuttx/configs/Kconfig
[see] (https://groups.yahoo.com/neo/groups/nuttx/conversations/topics/12415;_ylc=X3oDMTM3a2JjNWVmBF9TAzk3MzU5NzE0BGdycElkAzIzMzg5MDcwBGdycHNwSWQDMTcwNTAwNjU1OQRtc2dJZAMxMjU0NQRzZWMDZnRyBHNsawN2dHBjBHN0aW1lAzE0NjcyMjMyMzEEdHBjSWQDMTI0MTU-) for a discussion but you did not work from nuttx_v3 so you can not use the new solution.

This is way I suggested to use meld to but back thing like CONFIG_ARCH_BOARD and freinds

jmcwil, 2016-06-29:
@davids5 Thanks for taking the time to explain this to me!  I am doing my best to understand why all this works, and your explanation really helps.

Unfortunately, after successfully running all of the commands to set up kconfig and rebuilding the build_aerocore_default directory, `make oldconfig` is still giving an error.

`kconfig-conf: error while loading shared libraries: libkconfig-parser-3.12.0.so: cannot open shared object file: No such file or directory
make: *** [oldconfig] Error 127`

davids5, 2016-06-29:
See if this buids
```
  To suppress the graphical interfaces, use static libraries, and disable
  creation of other utilities:

    ./configure --disable-shared --enable-static --enable-mconf --disable-nconf --disable-gconf --disable-qconf --disable-nconf --disable-utils
    make
```

If it does then run     ```sudo make install```

If it does not it will tell you what things are missing ->packages need to be installed.

```While(not built) {
  google "thing that failed"
  apt-get or yum "thing that failed"
}```

jmcwil, 2016-06-30:
@davids5 While that ran, it still wouldn't let me use menuconfig.  However, I ran
`./configure --prefix=/usr --disable-shared --enable-static --enable-mconf --disable-nconf --disable-gconf --disable-qconf --disable-utils
`make`
`sudo make install`
`libtool --finish /usr/local/lib`
and that did the trick.  I think the last command is what I had been missing.

Thanks for your help! Menuconfig is a neat tool.

davids5, 2016-06-30:
@jmcwil - Glad to see you are making progress!

jmcwil, 2016-06-30:
@david5 Thanks!  I do have a couple of questions about this though, now that I am looking at this.  When I am in my build_aerocore_default directory, it appears that the changes I have made using menuconfig were applied.  However, is there something that I must do to the defconfig file in nuttx-configs/aerocore to make it reflect these changes?  If I clean my directory, what happens to the .config produced by menuconfig?

Additionally, I posted earlier about problems with an undefined reference to `io_timers` in `/drivers/stm32/libdrivers__stm32.a`.  i have looked in drivers/boards/aerocore to try to solve this file. I noticed that aerocore_timer_config.c was not in the CMakeLists, while the corresponding one for px4fmu-v2 was.  However, I tried including this file, and it would not compile.  Is this the file that will fix the problem with the stm32?
I found a comment in drivers/stm32/drv_io_timer.h that says "supplied by board-specific code".  I am assuming that the problem with io_timers that I am having is due to missing "board-specific code" for the aerocore, but I am not sure where to look to find this code.  Please let me know if I am on the right track, and where I would find the relevant code.

jmcwil, 2016-06-30:
UPDATE: Figured out the io_timers problem.  Turns out there were a few small typos in the files I mentioned.  I fixed them and now I seem to just have one more error that is keeping this from compiling: 
````CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj:(.rodata.g_builtins+0x2ac): undefined reference to `serdis_main'
CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj:(.rodata.g_builtins+0x2bc): undefined reference to `sercon_main'
collect2: error: ld returned 1 exit status````
If you know where it is, please let me know.  In the meantime I will keep looking around.  This has definitely been a great learning experience and I am starting to know my way around the code a bit.

LorenzMeier, 2016-06-30:
@davids5 Can you help out with pointers for the built-in apps?

jmcwil, 2016-07-05:
Hello all,

Thanks to everyone's help, I have successfully gotten the code to compile.  The last error was presumably caused by a directory that had changed names, so once I changed builtin_cmds to builtin_commands in my cmake file, everything compiled.

However, I still have work left to do.  While the code compiles, it will not upload to my device.  I have a feeling that it is still looking for the wrong bootloader.  I am getting this message:
`Loaded firmware for 13,0, size: 794976 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.` But plugging in the aerocore does not have any effect. 

I have succeeded in changing the number after "Loaded firmware for" from 9 (px4fmu-v2_default) to 13. My guess is that this is still the wrong number, and I believe the correct number may be 62 (I have code from the aerocore manifest that uploads with this number, although it does not work once uploaded).  Does this number tell the host computer which device I want to upload to?  How can I change this number to match my device?  Please let me know whether I am on the right track, and anything else I may be missing.

Thanks,
Jessica

davids5, 2016-07-06:
@jmcwil 

Great Progress! 

The next question I would as is:Did you build and JTAG the areocore [bootloader] (https://github.com/PX4/Bootloader/blob/master/hw_config.h#L330)  to your board? I am assuming the answer is yes or the bootloader came installed on the HW, because you have loaded FW with QGC - Is this correct?

Also what host PC/distro are you building and uploading the FW from?

jmcwil, 2016-07-06:
@davids5 
Thank you!

I have just been using the bootloader that comes preinstalled on the board.  Correct, I have successfully loaded firmware from QGC.  I have also verified that my host machine recognizes the board by running `ls /dev/serial/by-id` and getting the correct output.

My host is running Ubuntu 14.04.

davids5, 2016-07-06:
What do you have in the Images folder for the aerocore? Please post the file here.

davids5, 2016-07-06:
The bootloader for aerocore (1)  has 98 https://github.com/PX4/Bootloader/blob/master/hw_config.h#L340 for the board type.  So I would expect https://github.com/PX4/Firmware/blob/master/Images/aerocore.prototype should have board_id: 98

jmcwil, 2016-07-06:
@davids5 Here are the contents of aerocore.prototype in the Images folder.
`{
    "board_id": 19,
    "magic": "AeroCore",
    "description": "Firmware for the Gumstix AeroCore board",
    "image": "",
    "build_time": 0,
    "summary": "AEROCORE",
    "version": "0.1",
    "image_size": 0,
    "git_identity": "",
    "board_revision": 0
}`

Interesting, I agree that the board_id should be 98.  I am not sure where the 19 came from.  I will try changing it to 98.  Thanks for pointing me to this file!

jmcwil, 2016-07-06:
Thanks, the firmware is loaded for the right board now!  

Unfortunately, the board is still not quite responding.  Any ideas where I may have gone wrong?  I will keep putting files in meld to see if I can find something that looks off.

jmcwil, 2016-07-06:
@davids5 It looks like my menuconfig changes may not have stayed.  How can I make these permanent, so that running 'make clean' will not eliminate them?  When I used menuconfig previously, it only saved changes to .config in my build_aerocore_default directory. Not sure if this is causing my problem but it seems like something I should attend to anyway.

LorenzMeier, 2016-07-06:
You need to copy the resulting config to nuttx-configs/boardname/defconfig for it to stay.

jmcwil, 2016-07-06:
@LorenzMeier Thanks for explaining that!  I understand now, and I will remember to do that from now on.

Should I be seeing an aerocore_default.bin somewhere?  I do not seem to have one.  Please let me know if you have any ideas what may be causing the upload to fail.

LorenzMeier, 2016-07-06:
can you send a pr with your current state so i can run it?

jmcwil, 2016-07-08:
I was reviewing the pr, and I just noticed that it is failing Travis CI.  It appears that the most likely culprit is a problem with the aerocore defconfig.  I went through the process with menuconfig in the build_aerocore_default directory and then copied the resulting config to the nuttx-configs/aerocore/defconfig, but it is still possible that I made an error or have a wrong value in defconfig.  Perhaps I somehow broke a dependency?  Is it likely that such an error would prevent the upload from occurring?  Or is this more of a separate issue?  Please let me know, as I am trying to pinpoint the source of my problem.

Thank you!  I really appreciate everyone's helpfulness and responsiveness!

davids5, 2016-07-08:
@jmcwil - I am not sure if there is any real errors - it looks to be minor - there are some code style violations and the ```make check_format``  is catching it. run ./Tools/check_code_style_all.sh and will list the errors and tell you what to run to fix it.

jmcwil, 2016-07-08:
@davids5 Thanks for your response!  Running that script gives the output `Tools/fix_code_style.sh: line 3: astyle: command not found`
printed about 20 times, followed by
`Format checks passed`.
Not necessarily the output I was expecting, but seems ok to me.

Any speculations on what could cause the firmware to still not upload?  Even a list of which files are most critical to the process would really help me out.

Thanks again!

dagar, 2016-07-08:
Ha, that's not right. astyle is the checker. 
Try installing astyle and run `make check_format` instead. That check all script wasn't intended to be run directly.

davids5, 2016-07-08:
@dagar :) I run it all the time from the command line. LOL I  would request it be enhanced to have a -s option to not output the dif and produce pipeable output to run the fix or a -f option to do the fix.

jmcwil, 2016-07-11:
@dagar Ohhh I see now, thank you!  It looks like I just have some formatting errors- thinks like curly brace style and such.  Is it necessary to fix all of those manually?

dagar, 2016-07-11:
Nope, no manual fixing. Hopefully the tool told you what to do, but it's just

    ./Tools/fix_code_style.sh <FILE>

jmcwil, 2016-07-11:
@dagar That is much more convenient!  Thank you!

LorenzMeier, 2016-12-26:
Closing as stale.

closed 2016-12-26 by LorenzMeier