PX4/PX4-Autopilot #4828: Startup problem after Firmware 1.3.2 update

Namorus, 2016-06-15:
Hi
today I updated my firmware to 1.3.2. I used the source code, followed the guide http://dev.px4.io/starting-building.html and successfully built the project. After successful upload, my px4 got stuck somehow. 
Serial output yields the following:

**_sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param]    selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
WARN  [px4io]    CRCs match
INFO  [dataman]  Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst]      no devices found
INFO  [ver]      match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
LIS3MDL bad ID: 3cWARN  [lis3mdl]  no device on bus 2
WARN  [hmc5883]  no device on bus 1
ERROR [mpu6000]  no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
nsh: sf10a: command not found
INFO  [pwm_out_sim] MODE_8PWM_**

the red light of the arming button is blinking but it never stops. The big LED on the px4 never comes to light. Am I doing something wrong or is there an issue with the newest firmware release?

dagar, 2016-06-15:
If you power cycle does it get stuck at the same point every single time?

This is a bit of a leap, but can you either format your sdcard or try a fresh one? I had this problem once last week, but didn't really have time to investigate. After changing the sdcard it went away.

Namorus, 2016-06-15:
Ok I tried the same procedure with different SD cards but with no success - still the same serial output message as before

dagar, 2016-06-15:
What's your local dev environment, gcc version, etc? Have you built + flashed successfully before?

Can you try completely cleaning and rebuilding? Be sure to commit anything you've changed first.

    make distclean && make px4fmu-v2_default upload

Namorus, 2016-06-15:
I'm using OSX El Capitan and I strictly followed the [guide](http://dev.px4.io/starting-installing-mac.html) provided by the px4 team. Until now I have not changed something in the source code. Building and editing my code is done in Qt Creator (Qt Creator 3.6.1, Based on Qt 5.6.0 (Clang 7.0 (Apple), 64 bit)). The only thing not mentioned in the guide, which I had to install was Java SE Development Kit 8. 

**_maverick@Mavericks-MacBook-Pro:~$ gcc --version_**
**Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/c++/4.2.1
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin**

cleaning and rebuilding yield same serial out again:

**_sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param]    selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
WARN  [px4io]    CRCs match
INFO  [dataman]  Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst]      no devices found
INFO  [ver]      match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
WARN  [hmc5883]  ERROR: TIMEOUT 1
WARN  [hmc5883]  FAILED: SCALE
WARN  [hmc5883]  failed to enable sensor calibration mode
ERROR [hmc5883]  calibration failed
LIS3MDL bad ID: 3cWARN  [lis3mdl]  no device on bus 2
WARN  [hmc5883]  no device on bus 1
ERROR [mpu6000]  no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
nsh: sf10a: command not found
INFO  [pwm_out_sim] MODE_8PWM_**

With firmware version 1.3.1 I was able to build the project and even fly with my Quadcopter in HITL mode with jMAVsim.

dagar, 2016-06-15:
I think we've got a problem with pwm_out_sim introduced between v1.3.1 and v1.3.2.

LorenzMeier, 2016-06-15:
1.3.2 was actually a mishap, it went against master instead of stable. Please use https://github.com/PX4/Firmware/releases/tag/v1.3.3, which is what I intended to do. We are not supporting master for stable builds.

closed 2016-06-15 by LorenzMeier

dagar, 2016-06-15:
@Namorus this is resolved in current master and stable v1.3.3

PR #4829 (2016-06-15): pwm_out_sim sleep if no fds
-fixes #4828 

@julianoes I think you're familiar with this code. Can you review/comment? I don't fully understand why this is a problem at startup, but restoring the sleep resolves the issue in #4828
commit d16daf5b (2016-06-15): pwm_out_sim sleep if no fds (#4829)

-fixes #4828