PX4/PX4-Autopilot #4833: Takeoff possibly broken in SITL on stable branch

MHageH, 2016-06-16:
I used the general procedure, but instead of using the master branch, I tried to take off in SITL in the stable branch
```
$ git clone https://github.com/PX4/Firmware.git --verbose ; 
$ cd Firmware;
$ git checkout stable
$ git pull origin stable
$ git submodule update --init --recursive 
$ make -j4 posix_sitl_default jmavsim
```
The quad becomes unstable when off the ground. 
The takeoff is fine on the master branch tho.

TobiMiller, 2016-06-27:
Did you try autotakeoff command in reallife? i have the same behaviour there. 3 crashes and i thought it is my code. but now with stable stack as SITL i know it seems to be a firmware thing. I think this is critical for a stable firmware.

MHageH, 2016-06-28:
@TobiMiller Actually for the real hardware, I flashed the latest firmware provided from qgroundcontrol, not directly from the stable branch on PX4/master.
I directly use the C UART Interface to arm and fly without ever passing by auto takeoff or qgroundcontrol : [footage](http://sendvid.com/ziebmszp). 
Altho this happens on the stable branch, it doesn't on the master one.

LorenzMeier, 2016-06-29:
There is not enough context here - what does not work for you?

MHageH, 2016-06-29:
@LorenzMeier When you do a ``` make posix_sitl_default jmavsim ``` on the [stable](https://github.com/PX4/Firmware/tree/stable) branch, everything works fine, except when you try to do a ```commander takeoff``` from the NuttX shell, the quad crashes.
Is there any difference between the [stable releases](https://github.com/PX4/Firmware/releases), the PX4 [stable] (http://px4-travis.s3.amazonaws.com/Firmware/stable/index.html) firmware (used by QGroundControl) and the  [stable](https://github.com/PX4/Firmware/tree/stable) branch of PX4 ?

LorenzMeier, 2016-06-29:
No, those are all in sync. There will be a stable release within the next days.

MHageH, 2016-06-29:
@LorenzMeier Could you please check if you can reproduce this bug? It doesn't happen at all with the master branch.

dagar, 2016-11-08:
@MHageH is this still a problem?

MHageH, 2016-11-08:
@dagar I've test it again, the drone takes off to a stable position. You may close this issue.

closed 2016-11-08 by dagar