PX4/PX4-Autopilot #4836: SITL connection problem when running simulator in custom udp port

marianobianchi, 2016-06-17:
This bug affects both sitl_run.sh and sitl_multiple_run.sh.

If I edit the file "rcS_gazebo_iris" changing "simulator start -s" with "simulator start -s -u 15000" and run "roslaunch px4 mavros_posix_sitl.launch" I get no connection for mavros. However it works fine without changing the file.

The problem with this is that the relatively new way of launching multiple SITL instances use this variant of running different simulators in custom ports and I can't get a connection with them either.

I tried to figure out where is that port harcoded but I couldn't find it. I thought it was a bug in the way of setting the value of the port but checking the log I saw this message: "INFO  [simulator] Waiting for initial data on UDP port 15000. Please start the flight simulator to proceed.." so that is working fine.

dagar, 2017-04-11:
@marianobianchi did you ever figure this out?

marianobianchi, 2017-04-12:
I'm sorry, after reporting this issue I started working in another project and didn't try again to see how to fix this. I don't know if this is still an issue.

dagar, 2017-04-12:
Alright, thanks for the update. Please reopen if you'd like to pursue this.

closed 2017-04-12 by dagar