PX4/PX4-Autopilot #4837: hardfault in ekf_att_pos_estimator

Thezkiller, 2016-06-17:
I come across this problem,can anybody help me.

Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: ekf_att_pos_estimator
sp:     20002908
IRQ stack:
  base: 2000294c
  size: 000002e8
20002900: 2001ba40 0809c32d 080cffc4 000000b8 08097c49 00000010 20002654 00000003
20002920: 00000000 0809c5bd 0809c5a9 080a58d1 00000010 00000000 20019af7 006092da
20002940: 00000000 0809c52d 2001b8cc 00000000 10000010 00000001 00000000 00000000
sp:     2001b9a0
User stack:
  base: 2001ba40
  size: 000011f4
2001b9a0: 20019af7 20018750 00000001 0803e86b 20019a80 20019b0f 200197b8 ffffffff
2001b9c0: ffffffff 20019730 ffffffff 00000009 00000000 ffff0001 2001a31c 0000000c
2001b9e0: 00000000 00000101 2001a34c 00000000 00000000 00000000 00000000 00000000
2001ba00: 00000000 00000000 00000000 00000001 00000000 00000000 00000000 00000000
2001ba20: 00000000 0803e33b 2001a20c 00000001 00000000 0809911f 00000000 00000000
R0: 006094e7 00000000 000094e7 00000010 00000000 20019af7 006092da 00000000
R8: 00000000 00000000 00000000 00000000 00000000 2001b9a0 080b2fd7 080b2fd8
xPSR: 01000000 BASEPRI: 00000010 CONTROL: 00000000
EXC_RETURN: ffffffe9

in my ubuntu ,i tried this:
#arm-none-eabi-gdb nuttx-px4fmu-v2-default.px4
#info line *0x080b2fd8
#No line number information available for address 0x080b2fd8

dagar, 2016-06-17:
Which version of the firmware are you using? Did it correspond to anything, like shortly after getting a GPS fix for the first time (or connecting HIL)?

There was a hard fault in ekf_att_pos_estimator fixed about a month ago, but it hasn't made it to a stable release yet. It would occasionally be triggered the first time the ekf started publishing.
https://github.com/PX4/Firmware/commit/fe60a43bba6ebb976513f3db0b7a8aca749fc2a3#diff-691152f6db71750a66589c01dc9ef549

Thezkiller, 2016-06-17:
my version is v1.3.0rc3 .
commit ffb0d37c8a56c88e83e257b770053fe24946fc48 
Date:   Mon May 9 23:01:06 2016 +0200

dagar, 2016-06-17:
Okay so you definitely don't have that change, but that doesn't prove anything.

Can you try this?

    arm-none-eabi-addr2line -C -e build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx -a 080b2fd8

Thezkiller, 2016-06-17:
when i install my GPS module ,bug appeared.seems like the GPS cause the problem.
i try your command,
#arm-none-eabi-addr2line -C -e build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx -a 080b2fd8
#Firmware/src/modules/uORB/uORBDevices_nuttx.cpp:311

dagar, 2016-06-17:
Good, that lines up perfectly with the fix. The problem is/was the uninitialized mavlink orb descriptor, so when it tried to publish (uORB) you got the hard fault trying to access something bogus.

As a work around until the next stable can you cherry-pick this commit and rebuild? Otherwise we could potentially do a hotfix release.
https://github.com/PX4/Firmware/commit/fe60a43bba6ebb976513f3db0b7a8aca749fc2a3#diff-691152f6db71750a66589c01dc9ef549

Thezkiller, 2016-06-17:
thanks .the bug is fixed when i use your patch.

LorenzMeier, 2016-06-17:
This is fixed on stable a long time and the hotfix release is already out. Thanks for reporting!

closed 2016-06-17 by LorenzMeier