PX4/PX4-Autopilot #4838: make px4fmu-v2_ekf2 broken on master

ndepal, 2016-06-17:
I'm trying to flash a PixFalcon with PX4 master (bada390ddebb65562cb306fa0c98ee78ee5e3446). I want to use EKF2 so I do
```
make px4fmu-v2_ekf2 upload
```

The code compiles without issues and appears to successfully erase/program/upload.
After rebooting, the welcome tune sounds but QGC can't connect to it (connected via USB). I see the message

```
Your Vehicle is not responding. If this continues shutdown QGroundControl, restart the
Vehicle letting it boot completely, then start QGroundControl.
```

This does not help, however.

When flashing with
```
make px4fmu-v2_default upload
```
QGC connects as expected.

mhkabir, 2016-06-17:
Can you try setting this on the NSH console :

```
param set SYS_MC_EST_GROUP 2 
param save
```

And rebooting (while on the ekf2 config)

julianoes, 2016-06-17:
I've stumbled upon this as well in #4757, I'm still not really getting it.

ndepal, 2016-06-17:
@mhkabir I tried that, but QGC still can't connect.

gianni-carbone, 2016-06-17:
This should happen when a module cannot be loaded.
Is there something in the /etc/extras.txt (SD card) ?
try a default configuration (no extras.txt nor config.txt and no mixers on SD card)

mhkabir, 2016-06-17:
I think just booting with the serial console connection would reveal the
issue.

ndepal, 2016-06-17:
Looking at the SD card I can't see the /etc directory.
I did see a hidden folder with 'spotlight' in its name, i.e. a folder that my Mac must have put on there. After deleting that and the .Trashes folder I can once again connect to QGC.
I don't quite understand how files/folders on the SD card interfere only with a certain configuration.

gianni-carbone, 2016-06-17:
I suffers the same problem in my case there were a problem with a missing module in the start up process. 
You could customize the startup process with three files in the SD card:
http://http://dev.px4.io/advanced-system-startup.html

At this point you have to follow the mhkabir suggestion booting with serial console attached following the boot process messages.

gianni-carbone, 2016-06-19:
I've tried to compile and flash px4fmu-v2_ekf2 from master and I suffer the same problem.
Booting with serial console the problem is that estimator q cannot be load cause is an unknown command!
This is due the configuration in the system group was inav+aeq before flashing the ekf2 instead of default makefile.
To solve you can do the following:

flash px4fmu-v2_default
restart system
connect qgc
go to parameters/system
change and save SYS_MC_EST_GROUP to ekf2
flash px4fmu-v2_ekf2
remove power and reboot

the system work fine now.
This is a bug cause the ekf2 version firmware uses the latest SYS_MC_EST_GROUP value in case of previous default version flash
It is a more general problem that mavlink stop working in case of unknow command during the boot process.
hope help
gianni

ndepal, 2016-06-23:
@gianni-carbone Thanks, with that workaround switching to ekf2 works.

closed 2016-11-08 by dagar