PX4/PX4-Autopilot #4840: Support for building px4fmu-v2_default using clang/LLVM

zvonimir, 2016-06-17:
To be able to apply SMACK software verifier (https://github.com/smackers/smack/) on PX4, I have been looking to building PX4 using clang/LLVM. Initially, I got lucky with the `posix_sitl_default` target, which compiles with clang without any problems. That was a good start :).

After this initial success, I got totally stuck on trying the accomplish the same thing with the `px4fmu-v2_default` target. Clearly this one is much more complicated since it has to perform cross-compilation for ARM, and things break left, right, and center. I have invested half a day into this without much success. I used clang/LLVM 3.6.2.

Does anyone have any experience building `px4fmu-v2_default` using clang?
Is that path at all supported, or do you have any plans to support it?
Finally, would others be interested in having clang/LLVM be supported?

Thanks!

LorenzMeier, 2016-06-17:
We would be super interested! Given that most software compiles fine it should be generally feasible, but it likely is somewhat involved. Do you want to paste the first errors you're getting here?

zvonimir, 2016-06-17:
Well, I guess I have a more basic question to start with :). How do I enable clang when my target is `px4fmu-v2_default`?
Yesterday I came up with a very hacky way to accomplish this that I kind of don't want to ever see again :).

bkueng, 2016-06-20:
Last time I tried, cross-compilation with clang was a pain. But that was a while ago, so maybe things improved. You should be able to set it here: https://github.com/PX4/Firmware/blob/master/cmake/toolchains/Toolchain-arm-none-eabi.cmake
Then I think NuttX is separate, you can start from here: https://github.com/PX4/Firmware/blob/master/nuttx-configs/px4fmu-v2/nsh/Make.defs

mayanez, 2017-02-07:
I'd like to take this up. From what I could tell, PX4 uses newlib instead of the built-in NuttX math libraries. Also, how do the nuttx-configs Make.defs fit in with the .cmake files?

dagar, 2017-02-07:
What were you ultimately planning to do? Wouldn't working with PX4 in SITL provide the majority of the benefit and be far easier to work with?

mayanez, 2017-02-08:
I want to run some custom LLVM passes and benchmark them on a
microcontroller.

<https://github.com/PX4/Firmware/issues/4840#issuecomment-278173383>, or mute
the thread

dagar, 2017-02-08:
To repeat what @bkueng start here for changing the compiler - https://github.com/PX4/Firmware/blob/master/cmake/toolchains/Toolchain-arm-none-eabi.cmake#L27

PX4 cmake calls the nuttx build. https://github.com/PX4/Firmware/blob/master/cmake/nuttx/px4_impl_nuttx.cmake#L327

I'm not sure how difficult it will be to get nuttx building, but keeping gcc for nuttx and only building PX4 with clang might be an option.

mayanez, 2017-02-08:
So I was able to build PX4 with Clang. However, the main issue is code size. Using -Oz the binary can fit in 1MB of flash but the stack frames are too large causing the application to overflow. I'll try to play around with RAM to see if I can make it work, but apparently code size seems to be an issue plaguing Clang.

davids5, 2017-02-09:
Please use 

https://semaphoreci.com/px4/firmware/branches/master/builds/1041

```

Building sizes
B   text	   data	    bss	    dec	    hex	filename
   8374	      1	   4688	  13063	   3307	build_esc35-v1_bootloader/src/firmware/nuttx/firmware_nuttx
   8158	      1	   3576	  11735	   2dd7	build_px4cannode-v1_bootloader/src/firmware/nuttx/firmware_nuttx
   8776	      1	   3724	  12501	   30d5	build_px4esc-v1_bootloader/src/firmware/nuttx/firmware_nuttx
   8712	      1	   3684	  12397	   306d	build_px4flow-v2_bootloader/src/firmware/nuttx/firmware_nuttx
   7934	      1	   3664	  11599	   2d4f	build_s2740vc-v1_bootloader/src/firmware/nuttx/firmware_nuttx
   8176	      3	   3640	  11819	   2e2b	build_zubaxgnss-v1_bootloader/src/firmware/nuttx/firmware_nuttx
```
And 
```
Building sizes
B   text	   data	    bss	    dec	    hex	filename
 880984	   5096	  15152	 901232	  dc070	build_aerofc-v1_default/src/firmware/nuttx/firmware_nuttx
1302716	   5784	  20412	1328912	 144710	build_auav-x21_default/src/firmware/nuttx/firmware_nuttx
 780496	   3748	   9116	 793360	  c1b10	build_crazyflie_default/src/firmware/nuttx/firmware_nuttx
1202540	   5764	  20632	1228936	 12c088	build_mindpx-v2_default/src/firmware/nuttx/firmware_nuttx
 988516	   4872	  10076	1003464	  f4fc8	build_px4fmu-v1_default/src/firmware/nuttx/firmware_nuttx
 948336	   5768	  18084	 972188	  ed59c	build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx
 893932	   5768	  18084	 917784	  e0118	build_px4fmu-v2_lpe/src/firmware/nuttx/firmware_nuttx
1421960	   6060	  20468	1448488	 161a28	build_px4fmu-v3_default/src/firmware/nuttx/firmware_nuttx
1334376	   5840	  23828	1364044	 14d04c	build_px4fmu-v4_default/src/firmware/nuttx/firmware_nuttx
 766288	   5240	  15568	 787096	  c0298	build_tap-v1_default/src/firmware/nuttx/firmware_nuttx
```

As benchmarks to beat with master

mayanez, 2017-02-09:
Looks like there's enough memory to extend the stack size for the few tasks that needed it. I'll need to do a few tests to make sure it's stable though. Here are the statistics for the build sizes using Clang for FMUv2:

```
text	   data	    bss	    dec	    hex	filename
980564	   3428	  18088	1002080	  f4a60	build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx
921268	   3428	  18088	 942784	  e62c0	build_px4fmu-v2_lpe/src/firmware/nuttx/firmware_nuttx

```

I will send out a PR for others to review and test as well.

davids5, 2017-02-09:
@mayanez - 

Please keep in mind that:

 

1. The code size of some of the bootloaders can not exceed 8K
2. Using more stack in not beneficial on an embedded system 
3. It will need to support other CPUs (NXP kinetis the STM32F7...)
4. Re vetting all the 24+ HW targets is not a small task.

dagar, 2017-04-28:
Nearly done in https://github.com/PX4/Firmware/issues/4840

closed 2017-04-28 by dagar