PX4/PX4-Autopilot #4843: External Compass Orientation

sirPerna, 2016-06-18:
External Compass Orientation only shows up if I do all the calibration, restart QGC and go to sensors again. Does anyone have the same problem?

LorenzMeier, 2016-06-21:
@DonLakeFlyer sounds like a QGC issue or at least one requiring a coordinated effort to resolve. Can we sync via Gitter to resolve?

DonLakeFlyer, 2016-06-21:
Let me look. My guess is some sort of QGC optional param problem. Some params don't show up until after calibrate.

LorenzMeier, 2016-06-21:
We could introduce a star in the param request logic, e.g. CAL_MAG* to refresh the whole mag cal and rebuild the UI.

DonLakeFlyer, 2016-06-21:
It already does that. If it's not optional param, the timing of it may have changed maybe so the UI doesn't see it early enough.

DonLakeFlyer, 2016-06-21:
Here is what I did:
- Boot from clean SD card
- Flash beta firmware
- Set airframe, reboot
- Mag cal
- Look at params in QGC: IDs are set but rotation params are all showing internal
- Look at params using nsh shell: Rotation params all show internal  as well (**this shows bug is on firmware side**). QGC is working correctly.
- Reboot QGC, offset params now come across but rotations are still all internal
- Reboot board, now rotations params are correct

LorenzMeier, 2016-06-21:
Thanks, that's extremely helpful. Let me go digging.

LorenzMeier, 2016-06-27:
@DonLakeFlyer I've pushed a fix to master and at least my testing suggests its fixed - would you mind re-testing?

DonLakeFlyer, 2016-06-29:
Fixed

closed 2016-08-01 by sirPerna