PX4/PX4-Autopilot #4845: VTOL architecture around position control

AndreasAntener, 2016-06-19:
@LorenzMeier @tumbili @bkueng 
I've been working on better position control during VTOL transitions. Instead of writing a complete 3rd position controller (although for some parts this will still happen/make sense like for altitude control during transitions #4521), I'm changing the behaviour of the other controllers during the transition. It actually only affects the MC position controller at the moment: I'm telling it to stop care for position in forward transitions and only do breaking during back transition (velocity 0).

But the flow of control is a bit unfortunate currently since the position controllers don't have an interface where they listen to the VTOL controller, my only option is to do it via the vehicle status. Problem shown here in red:

[image]

So what we could do is also route the control mode (extended with new modes and maybe also position setpoints) through the VTOL controller:

[image]

Thoughts?

sanderux, 2016-06-23:
@AndreasAntener in your proposal, would the vtol also stop caring about altitude during back transition? if no: quadplanes tend to increase altitude if the mc pitches back to reduce velocity, if a decrease is commanded thrust will reduce and that can cause destabilization.
if yes: then you might hit the ground

or are you thinking a different method for decreasing velocity?

AndreasAntener, 2016-06-24:
@sanderux I'm only talking about horizontal velocity. Let's discuss this once I've put up a proposal.

dagar, 2017-01-02:
Shall we start fleshing this out?

AndreasAntener, 2017-01-06:
**There are 2 main problem areas:**
1. Handing over control from the end of the transition to the usual controller
2. Position control during the transition

**Main concern is:**
Separating responsibilities, which in practice comes down to making sure that the normal controllers don't need to be aware of the VTOL state.

To solve the above points we need clean interfaces. However, there is currently no consistent data flow from the VTOL controller to the others. @dagar I have also thought about using the setpoints for that.

sanderux, 2017-01-06:
I agree that using setpoint types would probably be the best approach. We also use that for TAKEOFF and LAND modes. I am unclear why the your diagram implies a VTOL POSCTL, it seems to me this should always be the FW POSCTL during transitions.

RomanBapst, 2018-03-05:
@dagar Here we go again :) The problem is still the same after all this time:

- the position controller is not aware of the transition and as soon as the backtransition is done the mc pos controller kicks in without any proper initialisation. In practice the vehicle will demand pitch up max from one instant to the other which you can imagine is not very nice.

- I think anybody who is not using reverse thrust during the transition (like @sanderux ) and thus reducing ground speed to a minimum until the transition is finished has exactly this problem. If not then they are probably shadowing it with some mc pos control parameter which is of course not good either.

dagar, 2018-03-05:
@RomanBapst yep, although I think we can already do a better job handling this particular case without architectural changes or making too much of a mess in mc_pos_control.

RomanBapst, 2018-03-06:
@dagar I'm pushing with @Stifael and @MaEtUgR to bring in #8933 so that we can have a flighttask which does a proper slowdown.

closed 2019-02-13 by stale[bot]