PX4/PX4-Autopilot #4846: px4fmu-v2_x Pixhawk - calling unknow command at the system startup hangs mavlink communication

gianni-carbone, 2016-06-19:
As in object, the mavlink communication hang in case of calling an unknown command during the system startup.
For example, in my case, mavlink hanged in the latest masters due of removing gpio_led module that is called in my custom extras.txt.
For example mavlink hang flashing _ekf2 with SYS_MC_EST_GROUP < 2 cause q estimator is called but not present (unknown command) during boot process.

Hanging mavling result in qgc cannot connect and asks to reboot completely the system with no same problem at the folliwing startups.

Should mavlink communication stay alive in the case of boot error, permitting qgc communication, and errors displayed in the qgc error console?

LorenzMeier, 2016-06-21:
In general an error would be better, but I'm on the fence wether this is necessary. If you screw up your Linux or Windows boot you can also end up having no UI visible to you. We might want to make the boot not abort on error though, so I will keep this issue open since we should try to improve things.

gianni-carbone, 2016-06-21:
Lorenz,
it seems that in despite of the error I cannot connect via qgc but I can still arm the quadcopter (I did the test indoor so just armed not flown). 
At the first this kind of behaviour was really difficult  to show the real problem. Simply a different pattern in blinking LED could help to show the problem. I say this cause beginner user not often have a serial converter to debug the system console - see issue #4838 and this way is very confusing.

Thank you for your attention. 
And thank you for the job you are doing is a very great work!

dagar, 2017-04-11:
I'm not sure if we can do much here other than working towards a system that doesn't need extras.txt

jlecoeur, 2017-05-23:
+1 
I just lost hours because of this. I used the file /etc/extras.txt to start a custom app, and I forgot to remove the file when switching to master (were obviously my app does not exist).

I agree @LorenzMeier that if you screw up your boot script you cannot expect things to work properly. 
However this is currently very difficult to debug: no error indicated on LED, and nothing on mavlink.
Even a boot abort would have been easier to debug as I would have directly looked into the init scripts. 

As a minimal solution, maybe we can add a paragraph in the documentation to describe the problem and warn new users?

@dagar are you saying this is not possible to detect during the boot that a command in extras.txt is unknown, to then display an error?

LorenzMeier, 2017-05-23:
@jlecoeur Can you please try the following:

1) Add `set +e` to the top of your NSH script on CURRENT master
2) Try adding an incorrect command to your SD card

It should not abort any more, just throw an error on the shell. If that works, can you send a PR? We gained this ability only recently.

jlecoeur, 2017-05-30:
@LorenzMeier your solution works, thanks! 
However I am not sure if you meant sending a PR to make `set +e` the default. I do not think that failing silently by default is a good solution because one may want to be sure a custom app is running before taking off.

I added a paragraph in the doc in PR https://github.com/PX4/Devguide/issues/166 to mention the problem, and the `set +e` solution.

I also added an LED error signal in case of failed boot in PR #7315, please let me know what you think.

LorenzMeier, 2017-08-20:
We changed this and this is no longer an issue.

closed 2017-08-20 by LorenzMeier