PX4/PX4-Autopilot #4850: FW: Cannot takeoff in mission mode anymore

tumbili, 2016-06-20:
In mission mode the prop should start spinning and rev up once takeoff is detected.
This did not work anymore with the last stable release of QGC on the tablet.

dagar, 2016-06-20:
Did you have a takeoff waypoint? Hand launch or runway?

tumbili, 2016-06-21:
@dagar It had a takeoff waypoint and the user tried hand launch.

LorenzMeier, 2016-06-21:
I can't reproduce this in SITL:

```make posix gazebo_plane```

[image]
[image]

LorenzMeier, 2016-06-21:
I'm closing this as I think its was something else - the user couldn't arm at all in any mode. Reopen if it persists.

closed 2016-06-21 by LorenzMeier