PX4/PX4-Autopilot #4856: Avionics Power Low, Refused Spektrum Bind Command,

JPluess, 2016-06-20:
dear development team, 

1. i have a power module that provides 5.19v. and this makes up for 4.9v for the inside of the px4, 
there is a feature that bloks arming? i wish to disable this feature, 

the website states the voltage ratings for:

    Normal Operation Maximum Ratings

    Under these conditions all power sources will be used in this order to power the system.

    Power module input (4.1V to 5.7V) [refers to the voltage coming into Pixhawk from the power module]

    Servo rail input (4.1V to 5.7V)

    USB power input (4.1V to 5.7V)

http://ardupilot.org/copter/docs/common-powering-the-pixhawk.html

https://18358179047304348765.googlegroups.com/attach/fb472b8d967cf/Bild%203.png?part=0.1&view=1&vt=ANaJVrHRHhPsKXYT_fKIjlJql-q-hJc1fALLW2GKZyHaE6HHFI79lPugZXKbd6Y0eNSodQEsPz_Zael4vmFqxusobZBO_grG-4C5OBx27gkKaDVNxDUtiOM

2. The bind command is refused, satelite receiver wont go into bind mode.  see image.

https://18358179047304348765.googlegroups.com/attach/3ddae628b1d90/Bild%203ss.png?part=0.1&view=1&vt=ANaJVrF7JmasQypao-_mxsz60P13vyPaLoZwtcj1ppifc2mSkS0kCraB6DlNzq0T06SEi7x4YFIiRp9xiQQMNdzYqCh8o0-Siqa1dGPxBMmgI4MfNX1sWok
can somebody take a minute to look at this question? thanks alot
joel

LorenzMeier, 2016-06-20:
Please flash current master (during the Firmware upgrade tick the advanced checkbox and choose the development version), re-do the airframe config (this will force you to calibrate sensors again) and report back. The power information you have was indeed based on outdated information. Better values are explained here: https://pixhawk.org/modules/pixhawk#voltage_ratings.

The voltage does not actually prevent you from arming. What other error message are you getting? Click on the exclamation mark.

Which autopilot are you using?

One last resort item you can try is to hold down the safety switch while applying power, which will force an IO processor reset.

JPluess, 2016-06-21:
[image]

[image]

well the bindcommand seems to be an issue on the beta version aswell, how about this. 

i have autopilot version 2.4.8

about the avionics power low error i will respond soon

LorenzMeier, 2016-06-21:
Your issue is what the autopilot tells you: The sensors are not set up correctly. Can you please re-do the airframe configuration and re-do the sensor calibration?

LorenzMeier, 2016-06-21:
We already have an issue for the sensor message: https://github.com/PX4/Firmware/issues/4695

The Spektrum binding works according to your output. You need to make sure to put your Transmitter into the right mode as well, as it will only pair if you do that on both ends correctly.

JPluess, 2016-06-21:
i have now flashed the development verison and spektrum bind is still not working. 
[image]

allright i have redone the sensor calibration, it wont arm, see message below, 4.87V, i know it does arm if i use a powermodule that gives a little more voltage, but taking it out, is difficult. 

the airfraime, i have set to this: 
[image]

LorenzMeier, 2016-06-21:
Are you trying to arm while you are on USB power? That is not allowed. Its possible that the low power message (which does not prevent you from arming, its just a warning) overlays the USB power warning. If you are on USB, try disconnecting the cable.

JPluess, 2016-06-21:
No actually i have a 3dr radio wired to the px4, and all commands are sent by 433mhz,

JPluess, 2016-06-21:
have you tried to arm one of yours on a adjustable voltage supply? so it should arm down to 4.8v? or even down to 4.1?

JPluess, 2016-06-29:
so what now? any news?

JPluess, 2016-07-07:
do you need any more information

agribov, 2016-07-25:
I have this issue as well. I'm powering my Pixhawk through a BEC on the servo rail, and about 50% of the time, when I try to arm it it gives me the "Avionics power low" shout-out, and the other 50% of the time it just arms. Usually it arms right after giving me the error.

Is there any way we can manually lower the avionics power threshold, or resolve this issue?

LorenzMeier, 2016-10-23:
We have since reworked this quite a bit. Early boards have a HW issue preventing bind, but in recent board everything works as expected with binding.

closed 2016-10-23 by LorenzMeier

1JonaB1, 2020-02-19:
I have the same problem any tipps ?

JPluess, 2020-02-21:
I had to change the powermodule