PX4/PX4-Autopilot #4857: Pixracer boots up already armed

CarlOlsson, 2016-06-20:
Flashed current master d1b27ab056d35f695b33d1a021d8900a4d69ba61 to my pixracer.
When I plug in the USB cable it boots up already armed. If I disarm from QGC I get the following warning message: "Avionics power low"
[image]

CarlOlsson, 2016-06-20:
changed airframe from "Generic 250 Racer" to "Generic Quad X config" and problems go away

LorenzMeier, 2016-06-20:
That actually starts the very same script. So I suspect the real fix was the param reset which the airframe change involved.

LorenzMeier, 2016-06-20:
Are you sure you didn't have the Snapdragon interface extensions loaded here? That would explain what you saw.

CarlOlsson, 2016-06-20:
I am not sure what you mean with Snapdragon interface extensions? I just changed the parameter CBRK_BUZZER to 782097 to disable the buzzer. The buzzer is still beeping but all my calibrations disappeared. (Radio, sensors etc)

CarlOlsson, 2016-06-23:
Flashed stable with QGC, configured "Generic Quad X config" and calibrated sensors, everything ok. Now I plugged in USB and connected again 30 min later, Pixracer boots up already armed. I disarm through QGC, constantly get "Avionics power low" and the airframe automatically changed to "3DR DIY Quad"

CarlOlsson, 2016-06-23:
Pixracer also boots up armed if I power it with 2s lipo battery. However, if I disconnect my RX and [pwm to sbus](http://www.readytoflyquads.com/pwm-to-s-bus-signal-converter) converter it does not.

LorenzMeier, 2016-06-23:
I'm pretty sure the arming issue is related to your S.BUS converter. The rest sounds like debugging noise and technically impossible.

CarlOlsson, 2016-06-23:
Yea I also believe that the converter is creating problems, also had super strange issues with APM. You don't happen to know any known issues with converters that can be fixed in SW? I have some last hours tomorrow to fix this

LorenzMeier, 2016-06-23:
Garbage in, Garbage out. I can only recommend to replace  your gear with something that actually works. A FrSky Taranis is incredibly cheap.

closed 2016-06-23 by LorenzMeier