PX4/PX4-Autopilot #4858: Unrecoverable free fall when thrust is set to 0 with offboard mode

zephyr834, 2016-06-20:
We are using a pixhawk for controlling a quadcopter using offboard mode in gazebo 6. We have run into an issue with freefall and setting thrust to 0. If the quad stays in freefall for more than one second we are unable to recover from the freefall, even when changing mode to auto_land. We are using the mavros topic /mavros/setpoint_attitude/att_throttle.

To replicate:
1. Set thrust to 1.0 and gain altitude to allow for a safe drop (700m is what we use)
2. Set thrust to 0.0 (setting thrust to a value that will cause it to not hover will present the issue)
3. Let it fall for a second. Freefall is detected.
4. Set thrust to 1.0 or change mode to auto_land. The drone will not recover and fall to the ground.

We have tried disabling freefall detection in the px4 firmware but the quad still goes into an unrecoverable state.

Additional notes:
- Setting the thrust to 0.0 and then back to 1.0 under one second typically allows the drone to recover. However, we cannot descend down at all in offboard mode.
- Our team has used jMavSim in place of gazebo and the issue is not present. We are able to save and recover the quad at any point of the fall.
- Ubuntu 14.04
- ros-indigo-desktop 1.1.4 from apt repo
- ros-indigo-mavros 0.17.1 from apt repo
- Latest PX4 firmware from master branch of github
- Using the topic /mavros/setpoint_position/local these problems do not occur if we move down.

We think its this issue: list issue
Is this working as expected? It doesn’t seem that way
We would appreciate help in identifying what the issue is and a potential solution (if there is one)

zephyr834, 2016-07-08:
Looking for any kind of response. Has anyone looked into this or even read this issue?

dagar, 2016-07-08:
Does it think your landed? Can you post the log http://logs.uaventure.com/?

closed 2016-11-08 by dagar